motioneye-project/motioneyeos #2198: Motion on two cameras simultaneously (without noticeable motion)

Nedjohh, 2019-11-15:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.
Y
I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.
Y

### motionEye Version (Running inside a Ubuntu VM)
0.41

### Camera
Ctronics POE Bullet-Cam
720P/1500kbps
Network Camera (RTSP)

### Network Connection
Wired, POE

### Peripherals
Running on Ubuntu 18.04 LTS (VMware 4GB. 3 Cores)

### Log Files
`[940488592:motion] [NTC] [ALL] conf_load: Processing thread 0 - config file /etc/motioneye/motion.conf
[940488592:motion] [NTC] [ALL] config_camera: Processing camera config file camera-2.conf
[940488592:motion] [ALR] [ALL] conf_cmdparse: Unknown config option "threshold_maximum"
[940488592:motion] [NTC] [ALL] config_camera: Processing camera config file camera-5.conf
[940488592:motion] [ALR] [ALL] conf_cmdparse: Unknown config option "threshold_maximum"
[940488592:motion] [NTC] [ALL] config_camera: Processing camera config file camera-3.conf
[940488592:motion] [ALR] [ALL] conf_cmdparse: Unknown config option "threshold_maximum"
[940488592:motion] [NTC] [ALL] config_camera: Processing camera config file camera-1.conf
[940488592:motion] [ALR] [ALL] conf_cmdparse: Unknown config option "threshold_maximum"
[940488592:motion] [ALR] [ALL] conf_cmdparse: Unknown config option "webcontrol_parms"
[940488592:motion] [NTC] [ALL] motion_startup: Motion 4.0 Started
[940488592:motion] [NTC] [ALL] motion_startup: Logging to syslog
[940488592:motion] [NTC] [ALL] motion_startup: Using default log type (ALL)
[940488592:motion] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
[1:ml1:Terrasse] [ALR] [NET] netcam_start: Network Camera thread starting... for url (rtsp://192.168.178.54:554/h264?username=admin&password=trf1eqtx)
[2:ml2:Vorderseite] [ALR] [NET] netcam_start: Network Camera thread starting... for url (rtsp://192.168.178.76:554/h264?username=admin&password=trf1eqtx)
[3:ml3:Rueckseite] [ALR] [NET] netcam_start: Network Camera thread starting... for url (rtsp://192.168.178.41:554/h264?username=admin&password=trf1eqtx)
[4:ml4:Eingang] [ALR] [NET] netcam_start: Network Camera thread starting... for url (rtsp://192.168.178.59:554/h264?username=admin&password=trf1eqtx)
[4:nc4:Eingang] [ALR] [NET] netcam_handler_loop: Camera handler thread [6] started
[2:nc2:Vorderseite] [ALR] [NET] netcam_handler_loop: Camera handler thread [7] started
[3:nc3:Rueckseite] [ALR] [NET] netcam_handler_loop: Camera handler thread [8] started
[1:nc1:Terrasse] [ALR] [NET] netcam_handler_loop: Camera handler thread [9] started`

Hello,

i have a problem with two out of my four cameras. MotionEye detects motion on these cameras without anything moving in front of the cams. On both cameras the counter (top right) is jumping from 0 to over 300000. And it always happens on both cameras exactly at the same time.

If there is actual motion at these areas the cameras are pointed to, it works normal. But over the day i get around 50-60 misdetections per cam. As mentioned always at exact the same time on both cameras.

Last time this happened, i removed both cams from motionEye and readded them. After that they work for months without problems. Yesterday i had to reset one of my cams (just grey picture). Now the problem is back.

Tried a lot of changes on motion-detection (Despeckle on/off, autonoise, manually to 25%, aso). Always the same problem.

Anyone any idea what is happening here ?

Thanks in advance !

hexordecimal, 2019-11-16:
try turning off tcp in /etc/motioneye/motion.conf..just uncomment it, hope that helps

```
webcontrol_interface 1
webcontrol_port XXXX
setup_mode off
webcontrol_parms 2
webcontrol_localhost on

netcam_use_tcp off
```

Nedjohh, 2019-11-18:
That did it ! Thanks a lot. Never worked so smoothly before :)

Nedjohh, 2020-03-29:
One week ago, i had to restart my vm running motioneye. Now this behavior is back.

I have no idea what might have changed. The "netcam_use_tcp off" is still in "/etc/motioneye/motion.conf". Two cameras are again firing motion where no motion is happening. Always at the same time.

I tried to install a new vm for motioneye, defining all cameras as UDP from the start this time.
But in that case, the third half of the picture is always distorted, and so also constantly firing motion. And of course, looks terrible.

Anyone any idea on that ?

Thanks in advance,
Andre

starbasessd, 2020-03-29:
Which VM? VMWare? VirtualBox, and which versions?
What are you using as the Host OS, The VM Host (versions, too, please),
along with version of motionEye.
What kind of resources set up?
Network set to Bridged or Nat?
Which cameras?
Wireless or all wired?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Nedjohh, 2020-03-29:
Thanks for the fast reply !

Its indeed VMware (2 Cores 4GB) for the vm running on Ubuntu 18.04. Host is also Ubuntu 18.04 LTS Server (16 GB).
Version of motioneye is 0.41.
Network is bridged. All components are wired.
The cams are 4 CTronics POE running at 720P 25 Frames ( h264).

Hope this helps. Thanks again.

starbasessd, 2020-03-29:
Cut frames back to 5-10, resolution to 800x600 (just for testing)
Can't help with VMWare tweaks, I run Ubuntu 19.10 Host on 8 core Ryzen
7-1700/32GB RAM/10TB storage with VirtualBox 6.1x
My VM Hub is running Ubuntu 19.10 server, 4 cores, 4GB, 50GB, .41, 4.2.2.
3 D-Link DCS-5020-L and a Generic USB connected dashcam (currently, have
several other cams I use for testing), total 20=30FPS across the lot, all
set to 800x600.
I have a hub running Ubuntu 19.10 running on a TFF Lenovo, and usually have
a PI set up for testing.
I assume you can view the cameras directly when the VM is having issues and
the stream is OK?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Nedjohh, 2020-03-30:
Yes, the stream is working fine while this happens. I can see the red frames around a piece of wall or the floor, where no motion happened.

So it seems TCP and UDP are somehow colliding and the "netcam_use_tcp off" has no effect anymore ?

Anyone else any idea on that ?