motioneye-project/motioneyeos #2200: Controlling PTZ through real buttons

KjellKriminell6, 2019-11-18:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version:  20191109

### Board Model

I am using the following board/model: Raspberry Pi 3.

### Camera

I am using the following type of camera: Network Camera.

My camera model is: Axis IP-camera.

### Network Connection

My motionEyeOS unit is connected to the network via: Wifi..

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 External USB Drive.

### Log Files

I consider the following log files relevant to this issue:

 None
 
Hi!
I would like to be able to change position of my PTZ-camera and trigger recordings via physical buttons. Perhaps action buttons?
Can this be done via joystick-shield @ GPIO or perhaps a game controller och some gamepad connected go the motioneyeOS?

hexordecimal, 2019-11-20:
Check this out https://github.com/ccrisan/motioneyeos/pull/2016

hexordecimal, 2019-11-26:
i figured this out as well with gamepad
https://github.com/ccrisan/motioneye/issues/1548