motioneye-project/motioneyeos #2202: DROPBOX:  Accessing the upload service failed: HTTP Error 409: Conflict!

Jaizan, 2019-11-20:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

I am running motionEyeOS version:   | motionEyeOS 20190911

I am using the following board/model: PiZero W

I am using the following type of camera: MMAL (works OK)

My camera model is: (https://www.amazon.co.uk/gp/product/B07BLS4L6L/ref=ppx_yo_dt_b_asin_title_o00_s00?ie=UTF8&psc=1)

### Network Connection:  WiFi (connects)

### Peripherals   NA

### Log Files.    Extract below.    The whole log file is huge.

1   I have an error with Dropbox.    "Accessing the upload service failed: HTTP Error 409: Conflict!"
2   Following advice in another closed thread, I tried switching off uploads and re-entering the data, with a new key and have the same error.
3   I tried a reboot, same error
4   I updated the MotionEye OS, same error
 
This PiZero W has been operating for weeks with google, but I needed to swap the cameras from a Pi4 to the Pi zero and vice versa.   Google has the known "Sign in with Google temporarily disabled for this app" error which prevents me getting a new API key, so I had to try Dropbox as an alternative.
I can connect to the PiZero and view the camera from my PC. 

Here is an extract from the log file.    The whole log file is huge, as this Pi has been running for weeks on google, therefore I just included an extract.

ERROR: failed to upload file "/data/output/Camera1/2019-11-20/11-20-37.jpg" with service dropbox: HTTP Error 409: path/malformed_path/
Traceback (most recent call last):
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 1103, in upload_media_file
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 103, in upload_file
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 726, in upload_data
  File "/usr/lib/python2.7/site-packages/motioneye/uploadservices.py", line 776, in _request
  File "/usr/lib/python2.7/site-packages/motioneye/utils.py", line 817, in urlopen
  File "usr/lib/python2.7/urllib2.py", line 154, in urlopen
  File "usr/lib/python2.7/urllib2.py", line 435, in open
  File "usr/lib/python2.7/urllib2.py", line 548, in http_response
  File "usr/lib/python2.7/urllib2.py", line 473, in error
  File "usr/lib/python2.7/urllib2.py", line 407, in _call_chain
  File "usr/lib/python2.7/urllib2.py", line 556, in http_error_default
HTTPError: HTTP Error 409: path/malformed_path/
2019-11-20 11:20:53: [motioneye]    ERROR: dropbox: request failed: HTTP Error 409: Conflict
2019-11-20 11:20:53: [motioneye]  WARNING: accessing dropbox failed: HTTP Error 409: Conflict

TheHaf, 2019-11-29:
I can confirm this error, I also get it on my Pi 3B with the current motionEyeOS version 20190911. Maybe something changed in the Dropbox API...

Benjamin7785, 2019-12-01:
...so can I. I used it successfully until middle of November. But when I used it today it didn’t work anymore. 
First, I thought it was an error by setting up the „path“ correctly. When I was fiddling around with it I noticed that by using / instead of \ the test-function returns „success“ but no images are being uploaded.

Benjamin7785, 2019-12-03:
Well, I just configured a new SD-Card using a spare Rpi2b, set up everything and for what ever reason Dropbox upload works again. 
I checked all accessible settings of the RPi3b device, on which the upload doesn't work, but I cannot see a difference. 
That makes me kind of crazy :-)

Does anyone know if there is any developer setting that restricts access to a verified device by checking mac-address when not used for a while?