motioneye-project/motioneyeos #2203: Cannot connect to motioneyeos-raspberrypi4 outside of network

jthivakar, 2019-11-20:
I am running motioneyeos-raspberrypi4-20190911.img.xz on a Raspberry pi 4 using an IR-CUT night vision camera attached.

It is directly connected to the internet via Ethernet.  I was able to connect and get the IP, log in to the browser and access all the settings.  My issue is that I want to able to access the camera outside of my network.

In order to do this, I set a static IP, went into my Modem settings and enabled port forwarding and created a rule.

How do I access the camera now?  Outside of my network I typed my home IP Address and then ":" the external port.  I still can't connect.  Not sure what I am doing wrong.

Now when my raspberry pi boots up.  I get the following...

Configuring wired network: 192.168.1.101/24
Setting statis gateway to 192.168.1.1: done
Setting static DNS server to 8.8.8.8: done
Setting current date using http: failed
Starting http date updater: done
Starting crond: done
Starting sshd: done
Starting proftpd: done
Starting smbd: done
Starting ****: done
Starting motioneye: done
Interface eth0 has IP address 192.168.1.101/24
Default gateway is 192.1**.*.*
DNS server address is 8.8.8.8

Welcome to meye-#$%@#$%@#$%

pug-life, 2019-11-21:
You need to connect to the external public IP address you have (after you have set up the port forwarding correctly).
You can confirm that you have a public IP if the public IP your modem states is not in one of these ranges:

10.0.0.0 – 10.255.255.255
172.16.0.0 – 172.31.255.255
192.168.0.0 – 192.168.255.255

If your modem's WAN IP address is in the range of any of the above, you have a private IP address assigned from your ISP and it is not possible to reach it from the external network. You may contact your ISP to address this.

jthivakar, 2019-11-22:
Hi pug-life,

I used the static IP within the range.  I am able to log in using the new static IP within my network.

I use my public IP (which is not constant) and ":" port number
This still does not load the page.  The public IP assigned by my ISP is not in any one of those ranges listed above.  It starts at 75.**.*.*

When setting the rule on my Modem/router provided by the ISP.  I entered the same port for internal and external.

Not sure what step I am missing but I still cannot connect from outside my network.

Stofferl, 2019-11-22:
Some Providers are blocking several ports.
I think it is the best you call the support hotline of your provider and ask them.
Maybe you get support configuring your modem. They know your model and if it's possible.

jthivakar, 2019-11-25:
@pug-life @Stofferl 

 I tried a different port and it worked.  I believe the port was blocked by my ISP.
It would be nice to know which ports are being blocked.  Will need to contact them.

Thank you for all the help guys, appreciate it!