motioneye-project/motioneyeos #2204: Iphone shows "Accessng network share failed: Cannot mount network share!

Pszoom, 2019-11-21:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: (insert your version here, e.g. 20180314).

### Board Model

I am using the following board/model: (insert your board model here, e.g. Raspberry PI 3B+).

### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

My camera model is: (insert your camera model, e.g. Logitech C290).

### Network Connection

My motionEyeOS unit is connected to the network via: (choose your network connection, e.g. WiFi, Ethernet).

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)
 
Hi Calin, I am using a RaspberryPi 3 with a V2 camera module for Motioneye OS. I am using a RaspberryPi Zero W with a USB Hard Drive as a NAS for storage for the Motioneye setup.  It seems a lot of users have the same problem, "Accessing network share failed: Cannot mount network share". The software works great but when I try to setup the network share the camera stops working. I can map the NAS to my Windows 10 PC and copy and paste files to it and send files to it. The one thing I noticed when I try to setup the network share nothing happens when I click "Test Share" on the Windows PC but when I click "Test Share" on my iPhone I get the message "Accessing network share Failed: Cannot mount network Share"? I am just trying to setup V2 camera module as surveillance camera and would like to add more cameras later. Need to get one working first. I am new to all of this so I have attached a bunch of pictures and file for you to look at. A picture is worth a thousand words. Thank you for taking the time to read this. I hope the files moved over ok.

[motioneye.log](https://github.com/ccrisan/motioneyeos/files/3876906/motioneye.log)
[image]

[image]
[image]
[image]

[image]
[image]
[image]
[image]
[motion.log](https://github.com/ccrisan/motioneyeos/files/3876907/motion.log)

Pszoom, 2019-11-23:
So if  I take a picture manually I can download it from my Motioneye camera and "save as znasshared(\\RASPBERRYPI)(Z:)

hexordecimal, 2019-11-26:
i dont understand all those pictures but from what i can see with the iphone app the root directory is on /, where as the network share is /media/pi/NAS