motioneye-project/motioneyeos #2209: Wifi configuration not working on RPi3 Model B

sergiomafra, 2019-11-24:
#### Specs
SBC: Raspberry Pi 3 Model B
OS: MotionEyeOS v20190911
Camera: CSI with nightvision

Everything is working good so far, except on WiFi

I've read the wiki on how to configure wpa_supplicant. Tried without reflashing, tried reflashing and it doesn't seem to connect to my home network via WiFi.
I tried to configure the wifi network through the surveillance page without success either.

Is there any log I should copy and paste here so someone can help me? Is this a known OS issue? I looked through all issues trying to find something similar and haven't found any clue.

M-Abozaid, 2019-11-26:
I'm having the same issue 
also wired connection stops working if you try to setup the wifi through the web interface

I'm getting 
```
Starting wpa_supplicant: failed (brcmfmac)
Configuring wireless network: no link
```

my /boot/wpa_supplicant 
```
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country=EG

network={
    ssid="wifi ssid"
    psk="wifi password"
    key_mgmt=WPA-PSK
}
```

vitor-rossi, 2019-11-27:
Sergio, the reason your data folder is empty is because you have not mounted the partition into it. When you flash the OS onto your SD card it creates two partitions, and after you run the OS once it creates a third partition. This third partition (/dev/sdb3 in my case) will be mounted into /data when the Raspberry Pi boots. So mount this partition somewhere and edit /etc/wpa_supplicant.conf within this partition. Hope this helps, boa sorte.

sergiomafra, 2019-11-27:
Vitor, thanks for the tip. Unfortunately I've just tried that and it still not working :(

vitor-rossi, 2019-11-27:
Damn, OK I thought that would fix the issue. Another thing to try, if you haven't already, is to flash the OS with the -n option, which sets up the network configuration for you:

`sudo ./writeimage.sh -d /dev/sdb -i "motioneyeos-raspberrypi3-20190911.img" -n 'network:password' -s "192.168.1.100/24:192.168.1.1:8.8.8.8" 
`

Obivously, cd to the directory where writeimage.sh is located, replace /dev/sdb with you device name, replace the image string with where your OS image is located, and network:password to your values in this format. The -s option sets up the device to have a static local IP, not sure you'd need it. This is how I flashed the OS to my Pi 3B and everything worked first try. If this doesn't help I am out of ideas, hopefully someone else can pitch in.

digiltd, 2019-11-28:
Just checking some basics, you are trying to connect to a 2.4GHz not a 5GHz?

The **Pi 3** is not dual band so will can only connect to 2.4GHz, the **Pi 3 B+** is dual band and can do both.

sergiomafra, 2020-02-28:
Hey, Vitor! This way worked. I also changed the static IP because the DHCP range is from 192.168.0.2 to 254.

Sorry to take months to respond, but my life changed a bit since the last time I went here haha

Thanks again.

closed 2020-02-28 by sergiomafra