motioneye-project/motioneyeos #2211: Can't load Pan Tilt preset with crontab?

alphanumeric007, 2019-11-27:
Yes, I confirm that I have read the CONTRIBUTING guide before opening this issue.
Yes, I confirm that I have read the FAQ before opening this issue.
motionEyeOS Version is 0.41
I am running motionEyeOS version: 20190911.
I am using the following board/model: Raspberry PI 3A+.
I am using the following type of camera: MMAL,
My camera model is: Raspberry Pi V1.3
My motionEyeOS unit is connected to the network via: WiFi.
I am using the following peripherals that I consider relevant to this issue:
Pimoroni Pant Tilt Hat Full kit.

I have the Pimoroni Full Pan Tilt setup and working. Incremental Pan Tilt and 3 Presets.
When I plug in my power supply to boot up the Pi the servos on the Pan Tilt move slightly.
What I'd like to do is have it always move to one of the presets on boot up.
My preset3_1 file is in /data/etc and my preset3.py file is in /data/etc/python-scripts.
From putty I logged in as root and ran EDITOR=nano crontab -e
Then entered the following
@reboot /usr/bin/python /data/etc/python-scripts/preset3.py
ctrl o, enter, ctrl x
Doesn't work though?
@reboot python /data/etc/python-scripts/preset3.py
Doesn't work either.

Can somebody tell me what I did wrong?
I was following this guide
[https://github.com/ccrisan/motioneyeos/wiki/Tweaks#running-commands-at-regular-intervals-cron-support](url)

alphanumeric007, 2019-11-27:
I figured it out a way to do it.
I logged in as root with putty
I then ran nano /data/etc/userinit.sh
add the following line
/data/etc/preset3_1
ctrl x, y, enter.
preset3_1 is preset 3, the one I wanted to use.

closed 2019-11-27 by alphanumeric007

popoviciri, 2019-11-27:
Good find @alphanumeric007, I used this in my setup as well. I didn't even think I could actually position the camera at startup. Thanks!!

alphanumeric007, 2019-11-28:
If you put these two lines in that same file the Power and Status LED's will turn off. My pan tilt is in a window up close to the glass. And I can see the reflection of the LED's in the glass in my video feed.
You will still get activity on boot up and boot down, but once Motion Eye starts up they will turn off.

sh -c 'echo 0 > /sys/class/leds/led1/brightness'
sh -c 'echo 0 > /sys/class/leds/led0/brightness'