motioneye-project/motioneyeos #2212: Is there a change we will ever see HomeKit/HomeBridge Support build into MotioneyeOS?

ctschach, 2019-11-27:
I love MotioneyeOS for this simplicity of setting it up. Is there a chance we will ever see HomeBridge build directly into MotioneyeOS. It would be so convenient to add a camera directly to the home app on your phone without any additional computer in between to bridge the gap.

backfire800, 2020-10-19:
Hi MotionEyeOs Team, 
I would also love an  Apple Homekit integration in this os. What do you think? Is it possible.

Have a Nice Day
Bernd

starbasessd, 2020-10-19:
Shouldn't it be the other way around? Does Homekit support Webcams?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

backfire800, 2020-10-20:
Hi starbasessd,
you brought up an idea. I have a second Raspberry in my network.  So I installed the Homebridge on the second pi and stream from motioned.
Thanks for your inspiration.

Bernd

starbasessd, 2022-01-19:
Closed as completed

closed 2022-01-19 by starbasessd