motioneye-project/motioneyeos #2214: RPi with MotionEyeOS loses network connectivity to random cameras.

smokeybandit, 2019-11-29:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: dev20191031

### Board Model

I am using the following board/model: Raspberry PI 3B+

### Camera

I am using the following type of camera: Network

My camera model is: Various Amcrest POE cameras.

### Network Connection

My motionEyeOS unit is connected to the network via: Ethernet via wireless AP

### Peripherals

I am using the following peripherals that I consider relevant to this issue:
N/A
### Log Files

I consider the following log files relevant to this issue:

N/A

I installed MotionEyeOS in August on a new RPi when I moved to a new house.  Previously I had used Raspian with MotionEye for the last few years.  I have six cameras set up, all Amcrest of varying models.  Since then, four times now I'll randomly lose network connectivity to one camera (always a POE camera).  Meaning not just the RTSP stream, but pinging, wget, etc.  The RPi simply can't see that IP anymore. I never had this problem just running MotionEye on Raspian.

Troubleshooting:
-I can access that camera's admin page from Windows via wifi.
-I can ping that IP from the AP machine.
-I can ping/wget that IP from another RPi that's also connected the same way via that AP.

It's tough to do any additional troubleshooting with the limited tools available in the MotionEyeOS build

There is nothing obvious in any log file.  The only way around it is to give a new IP to the victim camera.

smokeybandit, 2020-03-03:
This continues to happen.  It happened today with three cameras.  No other machine on the network, including other RPis, has any issue accessing the cameras.  My new workaround is to log into the camera, change the IP to something else and immediately change it back to the previous IP.  I can't fathom any reason why the MotionEyeOS RPI suddenly can't access these IP or why that workaround works.

starbasessd, 2020-03-03:
Can you try the latest 20200203?
Can you turn on Expert Settings, Network Link Watch, and Connectivity Watch (you can leave the defaults on) or set them to one of your cameras.
Let me know if this makes a difference.

smokeybandit, 2020-03-20:
Unfortunately none of this worked.  Reboots don't clear it up. It still remains that the rpi with MotioneyeOS on it loses connectivity to one or more cameras randomly and I can't even ping or wget the IP.  Meanwhile all other devices on the same network can still access that IP.  My previous workaround required me to change the IP of the device but I let one go and after several hours, MotioneyeOS was able to see it again.

starbasessd, 2020-03-20:
By any chance, do you have any of the network monitors turned on?
Do you have "Fast Network" turned on any of the cameras?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

smokeybandit, 2020-03-20:
I have the Network Link Watch turned on but no other network monitors. 

I assume you mean the Fast Network option under Expert Settings?   Mine are all Amcrest so I assume that wouldn't apply.

starbasessd, 2020-03-20:
If you can SSH into the pi, or have a monitor aand keyboard attach, can you
run the following commands and report the results, please?
/usr/bin/vcgencmd measure_temp  # gives GPU temp
cat /sys/class/thermal/thermal_zone0/temp  @gives CPU tremp if you divide
by 1000

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

smokeybandit, 2020-03-20:
/usr/bin/vcgencmd measure_temp - 64.5C

cat /sys/class/thermal/thermal_zone0/temp  - 64452

starbasessd, 2020-03-20:
You are at the upper limit of temps. Things will start throttling, causing
lots of issues.
Check thread for #2318. I posted a way to log the temps. You can modify the
temps that throttle, but you should seriously consider heat sinks and/or
fans.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

smokeybandit, 2020-03-20:
Thanks, that makes sense.  I'll take a look at 2318

Chiny91, 2020-03-21:
This set me looking as I have the "vanishing IP cam" problem, although not a temperature issue it turns out.   For @smokeybandit 's R Pi 3+, knowledgeable folk say the [throttling starts at 60ºC](https://www.raspberrypi.org/forums/viewtopic.php?t=217056), and that is an undocumented change.   One or two workarounds are suggested, for some situations.

@smokeybandit; do let us all know what happens.

smokeybandit, 2020-03-23:
I installed the two default heat sinks that come with the "deluxe" rpi kits.  Temps now at 54C.  Still the same problem is happening.  Cameras start "vanishing"

ccrisan, 2020-03-23:
@smokeybandit could you please attach your log files, even though you say there's nothing obvious there?

smokeybandit, 2020-03-23:
[boot.log](https://github.com/ccrisan/motioneyeos/files/4372110/boot.log)
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/4372111/motioneye.log)

Some inconsistent workarounds that don't include changing the camera IP include changing a setting w/in motioneye (like a camera FPS) or rebooting multiple times.  In this log you'll see camera 1 failures, but at the same time, camera 2 and camera 4 also were offline.  In this case changing a setting multiple times did not fix it, but rebooting twice did.

smokeybandit, 2020-03-25:
It's happening again (not trying to pile on, just a chance to grab fresh stuff from the logs).

Nothing new in motioneye.log or boot.log.  Motion.log has a series of these errors:

[8:nc8:Driveway] [ERR] [NET] netcam_rtsp_handler_reconnect: Normal resolution: Reconnecting with camera....

starbasessd, 2020-03-25:
Can you please attach all 5 logs found under "Expert Settings"?
Also, you could turn on Debug for 24 hours, and include the files again.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

smokeybandit, 2020-03-25:
Here's the first batch.  I'll report back tomorrow after 24 hours of debugging
[motion.log](https://github.com/ccrisan/motioneyeos/files/4382297/motion.log)
[dmesg.log](https://github.com/ccrisan/motioneyeos/files/4382298/dmesg.log)
[boot.log](https://github.com/ccrisan/motioneyeos/files/4382299/boot.log)
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/4382300/motioneye.log)
[messages.log](https://github.com/ccrisan/motioneyeos/files/4382301/messages.log)

starbasessd, 2020-03-25:
Also, if possible, note when you notice missing cameras, what time it
happened.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

ccrisan, 2020-03-25:
@smokeybandit it would also be useful if you could tell us the approximate time of such a camera connectivity error; we'll have to look through a relatively large amount of debug data :)

smokeybandit, 2020-03-28:
I haven't had the issue since I turned debugging on (of course).  So please stand by.

starbasessd, 2020-03-28:
Isn't that the way, always?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

smokeybandit, 2020-04-10:
I hadn't had the problem in two weeks so I turned off debugging.  As soon as it rebooted, the problem started again.  I re-enabled debugging and the problem still exists, so here are the logs.

[motioneye (2).log](https://github.com/ccrisan/motioneyeos/files/4462569/motioneye.2.log)
[messages (1).log](https://github.com/ccrisan/motioneyeos/files/4462570/messages.1.log)
[boot.log](https://github.com/ccrisan/motioneyeos/files/4462582/boot.log)
[dmesg.log](https://github.com/ccrisan/motioneyeos/files/4462583/dmesg.log)
[motion (1).log](https://github.com/ccrisan/motioneyeos/files/4462584/motion.1.log)

starbasessd, 2020-04-10:
I see this in messages.1.log:
Apr 10 11:07:36 motioneye user.notice motion: [4:ml4:Backyard] [NTC] [NET]
netcam_rtsp_ntc: ******************************************************
Apr 10 11:07:36 motioneye user.notice motion: [4:ml4:Backyard] [NTC] [NET]
netcam_rtsp_ntc: The network camera is sending pictures in a different
Apr 10 11:07:36 motioneye user.notice motion: [4:ml4:Backyard] [NTC] [NET]
netcam_rtsp_ntc: size than specified in the configuration file.
Apr 10 11:07:36 motioneye user.notice motion: [4:ml4:Backyard] [NTC] [NET]
netcam_rtsp_ntc: The picture is being transcoded into the size
Apr 10 11:07:36 motioneye user.notice motion: [4:ml4:Backyard] [NTC] [NET]
netcam_rtsp_ntc: requested in the configuration.  If possible change
Apr 10 11:07:36 motioneye user.notice motion: [4:ml4:Backyard] [NTC] [NET]
netcam_rtsp_ntc: netcam or configuration to indicate the same size
Apr 10 11:07:36 motioneye user.notice motion: [4:ml4:Backyard] [NTC] [NET]
netcam_rtsp_ntc: to possibly lower CPU usage.
Apr 10 11:07:36 motioneye user.notice motion: [4:ml4:Backyard] [NTC] [NET]
netcam_rtsp_ntc: Netcam: 480 x 640 => Config: 640 x 480
Apr 10 11:07:36 motioneye user.notice motion: [4:ml4:Backyard] [NTC] [NET]
netcam_rtsp_ntc: ******************************************************

Do you have this camera rotated in Software?
If you disable/remove this camera does the problem still occur?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

smokeybandit, 2020-04-10:
I changed the resolution in motioneyeOS to 480x640 and the cameras reappeared. But I've had luck randomly changing settings and getting the cameras to appear before, so I don't know if changing the resolution actually "fixed" anything.

I'll monitor for a while.

smokeybandit, 2020-04-15:
No change.  The cameras still disappear for hours at a time.

starbasessd, 2020-04-15:
Short of replacing the PIs, I'm out of ideas.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Chiny91, 2020-04-16:
Drat, I was hoping this could be cracked 😞  Looking more carefully, for my 6 * cams on 3 * Pi, almost all the vanishing is down to one particular cam, so I guess I should test replace that one.   That is a big effort, so I'm just putting up with the problem for now.

smokeybandit, 2020-04-16:
Seems like it's some underlying issue with the buildroot image way deeper than motioneye. My limited networking experience can't find a culprit though.

smokeybandit, 2020-05-29:
This can be closed. I switched back to motioneye on Raspian and haven't had the issue since.  Thanks for looking into it.

Chiny91, 2020-05-29:
Really the issue has had a workaround, rather than a fix.    I'm about to swap R Pi, to see what happens with my vanishing cam, although sadly my new R Pi looks like it has gone missing in the post.