motioneye-project/motioneyeos #2215: Curious partitionning?

ouafnico, 2019-11-30:
Hi,

I'm using motioneye os on PiZeroW (last version 20190911).
I'm just curious about this partitionning I see on the system? Same partition mounted multiple times?

Filesystem                Size      Used Available Use% Mounted on
/dev/root               213.0M    190.6M      7.5M  96% /
devtmpfs                180.4M         0    180.4M   0% /dev
tmpfs                   184.6M     16.0K    184.6M   0% /dev/shm
tmpfs                   184.6M      1.6M    183.0M   1% /tmp
/dev/mmcblk0p1           29.9M     12.7M     17.3M  42% /boot
/dev/mmcblk0p3           57.3G     30.9G     23.4G  57% /data
/dev/mmcblk0p3           57.3G     30.9G     23.4G  57% /home/ftp/sdcard
/dev/mmcblk0p3           57.3G     30.9G     23.4G  57% /home/ftp/storage

Is it normal? 
Doesn't be risky?

Anyway nice work on motioneye.

Thanks

ccrisan, 2019-11-30:
@ouafnico the following partitions (that I suppose are the cause of your worries) are mounted multiple times as they are *bind* mounts:

    /dev/mmcblk0p3 57.3G 30.9G 23.4G 57% /data
    /dev/mmcblk0p3 57.3G 30.9G 23.4G 57% /home/ftp/sdcard
    /dev/mmcblk0p3 57.3G 30.9G 23.4G 57% /home/ftp/storage

They are used by the FTP server to allow access to your system. If you don't want that, just disable the FTP server in Expert Settings. Otherwise, I don't think there's anything risky about these mounts.

ouafnico, 2019-11-30:
I didn't see there are bind mounts, and the content is different on each mountpoint.