motioneye-project/motioneyeos #2216: Request: 12 hour time format for text overlay

Spokehedz, 2019-12-02:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue. **YES**

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue. **YES**

### motionEyeOS Version

I am running motionEyeOS version: **20190911**

### Board Model

**Raspberry Pi Zero W**

### Camera

I am using the following type of camera: **MMAL Camera**

My camera model is: **vc.ril.camera**

### Network Connection

My motionEyeOS unit is connected to the network via: **WiFi**

### Peripherals

**None**

### Log Files

**None**
 
Could we get the 12-hour '%i' in the text overlay? Or, really, more of the options from the 'standard' set of *nix date text formatting?

Metroidaron, 2021-01-30:
Just responding here because I was looking for the same solution; looks like I found a work around; you can use the custom text field with those values to create your own date format. :)

alexfornuto, 2023-03-14:
Hey, I'm looking for the same solution. Can you clarify "those values" into what specifically needs to go into the custom text field?

alexfornuto, 2023-03-14:
Figured it out. For the next Googler: If you want the same format as default but with AM/PM, what worked for me was `%Y-%m-%d\n %r` - no quotation marks.

starbasessd, 2023-03-14:
Use Custom Test, instead of the %H:%M use %r.
e.g.:
%d %m %Y %r
gives
14 03 2023 04:56:39 PM