motioneye-project/motioneyeos #2218: hostapd support and toggle ability within network settings

virtualjp, 2019-12-03:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

###
I have not included details of my setup as I am submitting this as a feature request.

Basically I'd like to implement a bit of a zero config for a new motioneyeos deployments (on Raspberry Pi boards).
As, in my case, I will primarily use motioeyeos devices wirelessly I have been toying with the idea of configuring hostapd to setup a hotspot on first build/boot of a fresh image, negating the need for connection to HDMI and allowing for a quicker and easier way to access the web GUI and enable WiFi connectivity.

I've successfully been able to do this by copying hostapd.conf, dnsmasq.conf and watch.conf to the SD card after writing the image and it boots up fine allowing me to connect my phone to it and further access the GUI so that I can then enable WiFi and enter my personal WiFi details (I want to avoide having to use wpa_supplicant.conf at image creation so as to make the process more versatile).
This all works fine, however although WiFi settings are correctly applied, hostapd kicks in after reboot and overwrites the DHCP assigned address on the adaptor, thus rendering me unable to communicate with motioneyeos on my personal network.
My feature request would be to add a toggle option under the network settings (Wireless) that allows for hostapd to be turned off, thus inhibiting the overwrite of the DHCP assigned IP.

Hopefully my request makes sense and how I have submitted it is acceptable.

As a footnote, I would just like to say I think this is an amazing project and I want to thank Calin Crisan and all the other contributors for their efforts on motioneyeos!

ccrisan, 2019-12-15:
@virtualjp being able to configure your system via a temporary AP mode makes a lot of sense and is definitely something to be considered. I'm not sure yet how this should be implemented but it's something that must be upstreamed to [thingOS](https://github.com/ccrisan/thingos).

jasaw, 2019-12-15:
I think it's a duplicate of https://github.com/ccrisan/motioneyeos/issues/1422