motioneye-project/motioneyeos #2219: Automatic shutdown after too many login attempts

Mohammed-Janahi, 2019-12-05:
Hello there, 

Before I begin I just want to say many thanks to @ccrisan and all those who contributed in developing this amazing CCTV solution. I using motionEyeOS as part of my school project to implement IoT solutions on three cameras. I have done my work exclusively on the GUI and have faced absolutely no issues whatsoever. 

However, since I am sure I'll be asked questions regarding security mechanisms, I was wondering if it is possible to configure (or introduce) the feature of having motionEyeOS shut down after a specified amount of failed login attempts. Better yet, the system will shut down AND delete all the media files currently stored on the system (could be optional). 

If the feature does exist and somehow I missed it, I apologize for opening up this issue but I would appreciate to know how can it be configured.

Thanks in advance!

LindsayFowler, 2020-03-28:
Well, deleting all your media files because of too many failed logins is not a good idea because then anyone could erase all of your recorded media.

As far as I can see though, MotionEyeOS admin and user access are completely vulnerable to brute force attacks where every possible password for each number of characters is tried.  

Indeed when I set the password incorrectly, I noticed that there was not limit to the number of password tries.. Oh oh.. and hey everyone knows that the admin account username is "admin".

Previously, Calin has said it is impossible to see failed logins - (https://github.com/ccrisan/motioneye/issues/307) but why is it impossible? But I do not see why: Surely if there is code that activates with a successful login, code can be activated with an unsuccessful login. 

Fail2Ban is a great bit of software that solves this problem for Raspbian and when I had a domain name pointing to my Raspberry Pi server I did indeed see thousands upon thousands of attempts to gain access from (supposedly) China. 

Valuing my privacy, I need a way to secure MotionEyeOS against brute force - after all it is only a matter of time - how much time only depends on the rate of attempts possible and the strength of password.

starbasessd, 2020-03-28:
> ccrisan/motioneye#307 <https://github.com/ccrisan/motioneye/issues/307>)

Comments:
1) Local and Cloud files can be erased, not FTP or SFTP via WebGUI if you
set the files to be write only (Yeah, you can do that to a folder on an FTP
server)
2) If you have Internet Access to the motionEye hub, it should be in a DMZ
network, not on your primary network.
You can then watch traffic to/from the motionEye system, logging, etc.
Yeah, motionEye isn't the greatest for security, but ...
Also, you do realize that ANY admin account can be vulnerable to brute
force, as most can't be locked out.
I would not expose the SSH port to the internet, only the stream or WebGUI
interfaces.
I might also sit just a motionEye Hub on the internet, no local storage,
just watching the streams.
The inside hub would do the heavy lifting (storage to S/FTP, etc) and the
DMZ only able to watch the streams. Small SD Card, in case someone turns on
recording, which will crash it if it gets hacked.
Traffic would be one-way out from internal motionEye to DMZ (controlled by
firewall) only, then from DMZ to internet.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

LindsayFowler, 2020-03-31:
If the attacker has admin access, hiding the storage behind a firewall is not really going to do anything - he could just configure motion eye to store data on his own server. 

I suggest improving Motion Eye OS security by slowing the number of retries and banning the IP of anyone that makes more than 5 attempts.

starbasessd, 2020-03-31:
If you ban the IP, because, heck, ever mis-type the password to something
and get locked out, and have to call IT? Who's going to be able to unlock
it, un-ban the IP, etc? It's going to be a re-image.
Who are you trying to prevent getting in?
A perp concerned about cameras? Physical security of the hardware comes
first.
A perp not at location (hacked into your network)? Firewall/DMZ and don't
expose the system to the Internet.
Who is going to recover a lock-out?
Differing security concerns have differing and sometimes conflicting
solutions. motionEye takes a middle of the road path and compromises at
both ends.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-03-31:
BTW, I could write a script once any machine on the network is compromised
to spoof all IPs on a network, and attack the admin account, thus banning
all network access. Trivial.
Most outside attacks are going to present from your router inbound, so
pre-block it. Oh, wait, that won't work, because YOU can't access it.
If the admin account can be locked out, then the machine is inaccessible
even from the console.
Pick your attack vector, and where it can come from, and harden against it.
This is not designed, nor hardened for, an Enterprise.
Most of the Enterprise solutions are sub-netted, DMZ'd, if using Active
directory or similar different domain access lists, layer upon layer upon
layer.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

LindsayFowler, 2020-03-31:
I think you may be deliberately making a mountain out of a molehill. Sure you can write a script but you first would need to compromise a machine. The purpose of a lockout is to slow down a brute force attack, from anywhere. The lockout could be lifted automatically after reboot or a specified time, or even an link from an email. Look this type of basic common sense security is used in website accounts ubiquitously. It doesn't take a security expert to see how it could be effective. I stand by the suggestion and invite positive ideas.

starbasessd, 2020-03-31:
Again, what is the actual goal?
Protection of data?
Don't need elaborate log in protections.
Protection of access?
Don't allow internet access.
IMO the system is great for what it was designed for. Low cost, compact
network camera control system able to run on $35 or less SBCs where I
wouldn't need or expect 'high security'.
Want those features? Source code setup is available for you to mod. If
enough people were to request it, I'm sure the developers would consider
it, after all, they did a lot of extra work to
get GDrive working.
Map out your 'ideal' scenario that you want to have happen. If x, then Y.
If it seems reasonable, heck, even I might write it up and submit it to the
developers.
IMO, there are better ways to harden the system, without touching the
system. (again, firewalls, DMZ's, etc, most of which are available on the
average home router...)

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

LindsayFowler, 2020-03-31:
The goal is a camera that can be accessed from anywhere in the world, but not easily hacked with a brute force approach. If you have a solution that you care to explain, I'm all ears.