motioneye-project/motioneyeos #2220: MotionEye without OS is better

Piteglio, 2019-12-07:
Last summer I installed Motioneye on Raspberry Pi 3 b+ with the help of Camera Pi and the whole thing worked wonderfully. 
Then I read the noterellas that Motioneye would be installed over an operating system to work better. So I installed Raspbian and above this Motioneye.
At this point things no longer worked.
Turn and spin, also following some of your advice, the Camera Pi provides a unique shade of color similar to the sands of mars.
Soon I will install Motioneye again alone without any operating system.

jasaw, 2019-12-08:
Are you saying motioneye + raspbian does not work, but motioneyeos works for you? Which platform is giving you red image?

Piteglio, 2019-12-08:
Raspberry Pi 3 B+ and Raspberry Pi Camera Board - Night Vision & Adjustable-Focus Lens (5MP).
This is the equipment. 
I repeat: before installing Motioneye over Raspbian the camera made the best.
In a few days I take the SD card, format and I only put Motioneye.
After, I’ll tell you what happened next.
The adventure continues.

mishu28nmv, 2019-12-11:
Hello. I have the RPi zero WH and RPi NOIR camera. I see color images but not the infrared images. Should I disable the led or leave it on for dark environment? Or is there another setting I should make?
I'm using this inside my 3d printer, and the leds from the printer are turning off after a period of time!

dimgod59, 2019-12-17:
hello, I tested the motioneye version without OS for the SSH command (on / off, pan tilt etc ...) on the other hand with the same hardware (RPI3 + + picam IR) I saw nothing at night unlike has the motioneyeOS version .... (both latest version)
I looked at the camera settings, they were the same.
For me MotionEyeOS> MotionEye in night vision.
Too bad the SSH access doesn't work on MotionEyeOS :( that would have been perfect :)