motioneye-project/motioneyeos #2221: Installing additional python packages (f.e. for a Telegram bot)

xhain, 2019-12-09:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20190911.

### Board Model

I am using the following board/model: RaspberryPi 2 Model B.

### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

My camera model is: Raspberry Pi Camera Module

### Network Connection

My motionEyeOS unit is connected to the network via: WiFi

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * USB WiFi Adapter

### Log Files

I consider the following log files relevant to this issue:

* not required

---

Dear motioneye team,

thank you for setting up this great software.
I was wondering if it is safe to install additional python packages? I was hesitating to install pip because I wasn't sure it might break the system.
To be more precise: I was thinking of adding a motion run command or talk back & forth with the system using a Telegram bot. Find some info on that here: https://ginolhac.github.io/posts/diy-raspberry-monitored-via-telegram/

Thank you for your time.

fabiosoft, 2020-06-11:
no info?

xhain, 2020-06-11:
@fabiosoft 
Maybe check here:
https://www.openhab.org/addons/actions/telegram/#actions
Or these issues:
https://github.com/ccrisan/motioneye/issues/301
https://github.com/ccrisan/motioneyeos/issues/289

closed 2020-06-11 by xhain