motioneye-project/motioneyeos #2222: clock gone back top 1/1/1970 in Pi3b vesrion 0.41 dev20191203

neil969, 2019-12-10:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version
vesrion 0.41 dev20191203
I am running motionEyeOS version: (insert your version here, e.g. 20180314).
vesrion 0.41 dev20191203
### Board Model
pi3b
I am using the following board/model: (insert your board model here, e.g. Raspberry PI 3B+).
pi3b+
### Camera
PS23 eye 
I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

My camera model is: (insert your camera model, e.g. Logitech C290).

### Network Connection

My motionEyeOS unit is connected to the network via: (choose your network connection, e.g. WiFi, Ethernet).

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)
 
The date has reverted back to the 1st of the 1st 1970 after the last update, and I can't seem to change it using the date command  - what happened and how can I fix it ( motion detection stopped working too)

neil969, 2019-12-11:
I ended up reflashing the SD card and going back to the "stable" release. at least that proved the issue wasn't the Pi  ;) 
Note that when the date was wrong, the software crashed & "hung" when I tried clicking on " check for updates"

neil969, 2019-12-31:
So this happened again  with Motioneye V .40
Motion V 4.2.2 OS version 20190427
Date is back to the 1/1/1970 - happened after a network error, but the pi talk perfectly to the network

jasaw, 2020-01-06:
How are you syncing time on the MotionEyeOS device? NTP or HTTP? If NTP, ssh into your device and run `ps`, see if `ntpd` is running. The process may have died. If you are using HTTP time sync, I don't know how to check.

mickrick1, 2020-03-06:
I have the same problem and have checked but cannot see ntpd running. How do I start it? Ntpupdate does not seem to be a valid command.
I understand MotionEyeOS is a subset or branch of Raspian but a lot of advice on here seems to be for people running Motion on Raspian rather than MotionEyeOS - am I correct in that interpretation? As such, when ssh'ing into my MotionEyeOS, there are only a limited number of commands available and I haven't yet found where they are detailed (purpose, syntax etc.).

starbasessd, 2020-03-06:
RPi's do not have onboard real time clocks that keep time between boots.
If you do not have a network connection, you can ssh in and use the 'date'
command.
I'll get the command for you shortly.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-03-06:
command to set time & date with no network available:
 date -s "19 APR 2012 11:14:00"

It has been requested to add an option in Expert Settings to add a Manual
option to the Web GUI.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

mickrick1, 2020-03-06:
Once again, bang on with your advice - cheers!

starbasessd, 2020-03-06:
Glad to help. Have a good one.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?