motioneye-project/motioneyeos #2228: Missing / Advanced settings

lassewr, 2019-12-18:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version
I am running motionEyeOS version: (20190911).

### Board Model
I am using the following board/model: (Raspberry PI 3B+).

### Network Connection
My motionEyeOS unit is connected to the network via: Ethernet.

#############

Hi I’ve seen this mentioned before. I have made a setup with a raspberryPi zero w for a motioneye cam setup and then a server/hub with motioneye. This works fine. They can be connected and i was ready to cheer.

My problem is the advanced settings, where I can specify where to send the recordings and so on. 
It does not show. I’ve read many places that I’m supposed to activate the advanced settings under “general settings”. But I can’t find it.

Any ideas to finish my setup?

The following screenshots are from a clean install, made without adding the wpa_supplicant.conf
I’ve tried different SD cards and 4-5 re-flashing the image to the cards. none of it solved the problem.

[image]
[image]

lassewr, 2019-12-29:
I think this problem is really weird. All tutorials and screenshots show a different interface than the one I see.  
Did I miss something obvious?

jasaw, 2020-01-06:
Did you log in as Admin?

ccrisan, 2020-01-06:
Actually what's been previously called "Advanced Settings" is now gone. All settings are always visible for admin user.

closed 2020-01-06 by ccrisan

lassewr, 2020-01-06:
But I was logged in as admin !??

I think I’ll try with a clean install and pay even closer attention to that part. I don’t think I’ve ever logged in as user... can’t see why I would du that right now.

ccrisan, 2020-01-06:
@lassewr what particular options are you missing?

lassewr, 2020-01-08:
I formatet both SD cards in the Raspberry pi zero W and Raspberry pi3... and did one of the installation guides again. It looks like everything works now. I’m  really not sure what happened. I did look in as Administrator, but must have

balkalabs, 2020-01-21:
I know this issue is closed but i had the same.
For the record, "fast network camera" was turned on in expert settings tab.

lassewr, 2020-01-22:
> I know this issue is closed but i had the same.
> For the record, "fast network camera" was turned on in expert settings tab.

Thank you. It might be the reason... it’s working right now and “fast network camera” is indeed turned off.