motioneye-project/motioneyeos #2229: Question: Raspberry Pi Zero W low FPS / low light issues

sts-ryan-holton, 2019-12-19:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

20190911

### Board Model

Raspberry Pi Zero W

### Camera

Official Raspberry Pi Camera NoIR V2

My camera model is: (insert your camera model, e.g. Logitech C290).

### Network Connection

Connected to Wi-Fi on 2.4G, password protected network

### Peripherals

No Peripherals to report here

### Log Files

N/A

### My Issue:

Hi, I've recently bought a Raspberry Pi Zero W, along with the official camera. I've connected everything up and all seems well. However, the FPS is really bad, and the low light settings are basic unless switching to fast camera.

FPS is most important to me, being able to have a smooth video recording, as the camera will be recording footage 24/7 in low light conditions.

Here are my questions:

1. What's the best settings that I should have, and what things can I turn off which would be affecting the CPU usage/performance.
2. Is there any plan to include ISO settings when fast camera mode is turned off, I'd wish that this was an option

sts-ryan-holton, 2020-01-19:
Anyone able to reply?

sts-ryan-holton, 2020-03-05:
/cc @ccrisan

starbasessd, 2020-03-05:
You may find 20200203 a bit better in movies / streaming as far as
brightness, but you are going to run into hard limits, hardware/cpu/gpu and
wifi network as for the framerate.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

ccrisan, 2020-03-09:
@sts-ryan-holton 

> 1. What's the best settings that I should have, and what things can I turn off which would be affecting the CPU usage/performance.

Resolution and framerate are the factors that would impact the CPU usage the most. Video encoding and motion detection come in second.

> 2. Is there any plan to include ISO settings when fast camera mode is turned off, I'd wish that this was an option

motionEye doesn't come with its specific video parameters. It rather uses those exposed via V4L2 and passes them onto Motion which does the same. So, if your camera driver exposes an ISO setting, it will be available on motionEye as well. Otherwise it won't.

HOD42, 2020-03-09:
Not strictly answering the question but have you tried slightly overclocking the PiZero?
I've had reasonable results with moderate/medium overclocking and upping the GPU RAM (as high as 384MB). Just watch the temperature (especially if it's in a case).