motioneye-project/motioneyeos #2230: unable to open video device - logitech c270 with raspberry pi B

zaggynl, 2019-12-19:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: (insert your version here, e.g. 20180314).
**motioneyeos-raspberrypi-20190911.img.xz**
### Board Model
I am using the following board/model: (insert your board model here, e.g. Raspberry PI 3B+).
**Raspberry pi B**
### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).
**V4L2**
My camera model is: (insert your camera model, e.g. Logitech C290).
**Logitech webcam C270**
### Network Connection

My motionEyeOS unit is connected to the network via: (choose your network connection, e.g. WiFi, Ethernet).
**Wired**
### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)
 
[motion.log](https://github.com/ccrisan/motioneyeos/files/3985319/motion.log)
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/3985323/motioneye.log)
[messages.log](https://github.com/ccrisan/motioneyeos/files/3985324/messages.log)
[boot.log](https://github.com/ccrisan/motioneyeos/files/3985325/boot.log)
[dmesg.log](https://github.com/ccrisan/motioneyeos/files/3985326/dmesg.log)

I've tried the following change:
per https://raspberrypi.stackexchange.com/a/57671
mount -o remount,rw /boot
nano /boot/cmdline.txt
reboot now

zaggynl, 2019-12-23:
Seems to work after changing the device name in /data/etc/camera-1.conf from /dev/video10 to /dev/video0

Found /dev/video0 by running motion manually.

jasaw, 2020-01-10:
@ccrisan I think this is caused by newer MotionEyeOS releases creating 3 bogus cameras by default: `/dev/video10` `/dev/video11` `/dev/video12`. Is there a way to detect which is the actual camera?

Benny1961, 2020-01-13:
I tried this and camera only works for a few minutes. I am running just motion on raspberry pi 3 b with Raspbian buster full.

lukebarousse, 2020-01-15:
I'm running into the same issues above as zaggynl.  I have a Pi B with a Pi camera (Camera 1) and Logitech C270 camera (Camera 2). Pi camera works no issues.  I can't get C270 to work properly.  I messed with the camera-2.conf file and changed device name to dev/video0 (was /dev/video10) and got the 2nd camera to work, but then the Pi camera wasn't gettting mounted properly.  So I reverted to the normal camera-2.conf file.

getcouchd, 2020-01-26:
I can verify changing the device to /dev/video0 in /data/etc/camera-1.conf fixed the problem :)

CRImier, 2020-02-14:
Same for me, had to change the camera from video10 to video0. (RPi camera V2 NoIR on Pi4), and I also have the video10-video12 devices.

rainkinz, 2020-04-04:
I have motioneyeos setup with a raspberrypi cam and a logitech c320. I had to change:
```
videodevice /dev/video10
```

to 

```
videodevice /dev/video1
```

In order for both cams to work.

Jiheffe, 2020-09-04:
Can't get this to work...
RPi Cam V2 on Camera1 works OK...
Added a Logitech C270 and it mounted on /dev/video10. Tried moving it to /dev/video0 and /dev/video1 by editing /data/etc/camera-2.conf and rebooting, but none of them work. 
Any suggestion to further investigate?

starbasessd, 2020-09-04:
either ssh in, or at console, check
lsusb <enter>
and/or
dmesg | grep USB <enter>
Looking to see if the Logitech registers correctly (assuming the Logitech
is the issue)
Do any errors show in the logs?
Which version of motionEyeOS are you using?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Jiheffe, 2020-09-04:
Hi,

I deleted my previous comment because I finally got it working by changing videodevice from /dev/video10 to /dev/video1 in /data/etc/camera-2.conf.

For some reason it didn't seem to work at first (despite a reboot) so I moved it to /dev/video0 and again back to /dev/video1 (allways rebooting in between), and finally it worked. I must have missed something ;-)