motioneye-project/motioneyeos #2231: How to change where files saved-Linux NOOB

theehmfic, 2019-12-19:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version 0.41

I am running motionEyeOS version: 20120911

### Board Model 

I am using the following board/model: Raspberry Pi4 B

### Camera

I am using the following type of camera: V4L2

My camera model is: Logitech C290

### Network Connection

My motionEyeOS unit is connected to the network via:  WiFi

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

I have 2 Raspberry Pi 4s managed by a PC server running Ubuntu 18.04, MotioEye version 0.41, Motion version 4.2.2

I have everything up and running correctly, no issues there. The question I have is, how do I change where the video files are saved to? I installed MotionEyeos and Motion according to the installation on the wiki, so it saves them to the SD card I believe. Under File Storage it says "Custom Path" Root Directory "/data/output/Camera1" It says that for both of the boards. I would like to save the video and photos to the internal 1TB hard drive I installed specifically for this. It is not the same drive as the OS. I am really new to Linux and don't understand the file structure and naming convention. I have explored the forums and internet but it's like basically reading another language. The 1TB hard drive has a name of /dev/sdb1. 
When I'm in motioneyeos and click the drop down for file storage I'm presented with 3 options, Custom Path, Network Shares and SL16G 0x1199cb850 (/dev/mmcblk0p3). I have no idea what that 3rd option means. If anyone has any insight into this and is able to explain to me how to change where the files are saved I would greatly appreciate it.

phyf3r, 2020-05-09:
this is also what I am having a hard time figuring out

starbasessd, 2020-05-09:
your drive will be found at /data/media/
as sda(x) or sdb(x) or similar
You can check by ssh in to PI and run command
df -h
in my case it shows
[root@meye-f7e06d23 ~]# df -h
Filesystem                Size      Used Available Use% Mounted on
/dev/root               213.0M    186.8M     11.2M  94% /
devtmpfs                369.3M         0    369.3M   0% /dev
tmpfs                   373.8M     16.0K    373.8M   0% /dev/shm
tmpfs                   373.8M      1.1M    372.7M   0% /tmp
/dev/mmcblk0p3            6.3G     31.2M      5.9G   1% /data
/dev/mmcblk0p1           29.9M     14.7M     15.2M  49% /boot
/dev/mmcblk0p3            6.3G     31.2M      5.9G   1% /home/ftp/sdcard
/dev/mmcblk0p3            6.3G     31.2M      5.9G   1% /home/ftp/storage
overlay                   6.3G     31.2M      5.9G   1% /usr
overlay                   6.3G     31.2M      5.9G   1% /var/log
overlay                   6.3G     31.2M      5.9G   1% /var/lib
/dev/sda1               252.0M     52.2M    199.8M  21% /data/media/sda1
/dev/sda1               252.0M     52.2M    199.8M  21%
/home/ftp/storage/sda1
/dev/sda2                99.3G      3.4G     91.8G   4% /data/media/sda2
/dev/sda2                99.3G      3.4G     91.8G   4%
/home/ftp/storage/sda2

and I would use the line
/data/media/sda2
and put in File Storage, Custom Path
/data/media/sda2/<name of folder>

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

pearbo, 2020-06-20:
Hi,

I am also struggling with this too. I have my df -h below.

Have mounted the large hard drive with,

/dev/sdb1       1.8T   77M  1.7T   1% /mnt/videofiles

I then change the directory in motioneye to /mnt/videofiles/anything 

I am unsure if its working as it still shows drive usage as my OS drive, 110gb. 

For information I have motioneye as an add on for home assistant supervised on a linux server.

Can anyone help?

NAME   MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
loop0    7:0    0    97M  1 loop /snap/core/9289
loop1    7:1    0  93.9M  1 loop /snap/core/9066
sda      8:0    0 111.8G  0 disk
├─sda1   8:1    0   512M  0 part /boot/efi
└─sda2   8:2    0 111.3G  0 part /
sdb      8:16   0   1.8T  0 disk
└─sdb1   8:17   0   1.8T  0 part /mnt/videofiles
root@linuxserver:~# /mnt/videofiles
-bash: /mnt/videofiles: Is a directory
root@linuxserver:~# df -h
Filesystem      Size  Used Avail Use% Mounted on
udev            3.9G     0  3.9G   0% /dev
tmpfs           787M  1.8M  785M   1% /run
/dev/sda2       110G   12G   92G  12% /
tmpfs           3.9G     0  3.9G   0% /dev/shm
tmpfs           5.0M     0  5.0M   0% /run/lock
tmpfs           3.9G     0  3.9G   0% /sys/fs/cgroup
/dev/loop0       98M   98M     0 100% /snap/core/9289
/dev/loop1       94M   94M     0 100% /snap/core/9066
/dev/sda1       511M  6.1M  505M   2% /boot/efi
overlay         110G   12G   92G  12% /var/lib/docker/overlay2/e3ffb125a78cfa3d821480e1a05d8d1c9edef86203b23ccbfa31a64f6ca5aed7/merged
overlay         110G   12G   92G  12% /var/lib/docker/overlay2/2914d0b98ec1b994e00443293eed5e9291d75f4bdc7092a6abd9217b43a1a785/merged
overlay         110G   12G   92G  12% /var/lib/docker/overlay2/42a6a552990e734ece4540a4239305213222c166e8dfe06c2f805a28aaf5eb3c/merged
overlay         110G   12G   92G  12% /var/lib/docker/overlay2/15c7a625d9c17f628902f389f676cb373fc4002379061d3643e1021a9c63117c/merged
overlay         110G   12G   92G  12% /var/lib/docker/overlay2/efe917f79d2c28d29209b194397194902dc1df2a9d08cc92853b4296989301be/merged
overlay         110G   12G   92G  12% /var/lib/docker/overlay2/efac99e351b36cc24c8f7f59407d1f217cff8b3b8765b974ad0e4474ffa164d6/merged
overlay         110G   12G   92G  12% /var/lib/docker/overlay2/2505eac7585da01df0b8c2dde0f8c7669ae3eb131c55faa23edc3a53abde2212/merged
overlay         110G   12G   92G  12% /var/lib/docker/overlay2/83b48683c9c4345e286730a8d950b25203eab2d4fa09fbb2d87b3ee40f885dda/merged
overlay         110G   12G   92G  12% /var/lib/docker/overlay2/77e74c05065eb9559a3ea0bfbbdad96add96c553cdf27d05c244db350dda9e16/merged
overlay         110G   12G   92G  12% /var/lib/docker/overlay2/0309d643938dca77735044ee73d58b5d8f2ae4f64405cec88b2c5838f0f4dc90/merged
overlay         110G   12G   92G  12% /var/lib/docker/overlay2/71c87675fe4eb7a9232f95bef83e382a680b3871353a8bf519e45d276bacb2ad/merged
overlay         110G   12G   92G  12% /var/lib/docker/overlay2/54c6ac661f852dd173c63d46adb88a607bf5f2c87a2b98edadc7964e86c0dd2c/merged
tmpfs           787M     0  787M   0% /run/user/1000
/dev/sdb1       1.8T   77M  1.7T   1% /mnt/videofiles

starbasessd, 2020-06-20:
Who owns the /mnt/videofiles?
run:
ls -la /mnt and paste here
Who is running the motionEye Application?
run
ps aux | grep motion

paste the results here.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

pearbo, 2020-06-20:
thanks for the quick reply.

results below.

root@linuxserver:~# ls -la /mnt
total 12
drwxr-xr-x  3 root root 4096 Jun 20 14:40 .
drwxr-xr-x 24 root root 4096 Jun 11 06:32 ..
drwxr-xr-x  3 root root 4096 Jun 20 14:34 videofiles

root@linuxserver:~# ps aux | grep motion
root      6304  0.0  0.0    204     4 ?        S    14:13   0:00 s6-supervise mo                                                                                                                                                             tioneye
root      6307 14.8  0.4  39300 32292 ?        Ssl  14:13  15:15 /usr/bin/python                                                                                                                                                             2 /usr/bin/meyectl startserver -c /data/motioneye/motioneye.conf
root      6363  0.0  0.2  29696 22112 ?        S    14:13   0:00 /usr/bin/python                                                                                                                                                             2 /usr/bin/meyectl startserver -c /data/motioneye/motioneye.conf
root     12713  174  2.8 328316 233240 ?       Sl   15:39  28:40 /usr/bin/motion                                                                                                                                                              -n -c /data/motioneye/motion.conf -d 5
root     12892  0.0  0.0  13136  1012 pts/0    S+   15:55   0:00 grep --color=au                                                                                                                                                             to motion

starbasessd, 2020-06-20:
Root owns /mnt/videofiles, Root is running motioneye. Shouldn't be an issue
to write there.
Let's try:
sudo chmod -R 666 /mnt/videofiles (change mode to read/write for root,
group root, and everyone) and reboot and test (This is ONLY a test, it's a
security risk, too, you don't want to leave in place long term)

How are you mounting the /dev/sdb1 to /mnt/videofiles? manually?
/etc/fstab? Can you paste your mount command here or the line from
/etc/fstab?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

pearbo, 2020-06-20:
I have tried sudo chmod -R.... and it appears to do nothing. Just requests another command.

Mount command is -     /dev/sdb1 /mnt/videofiles ext4 defaults 0 1

This is mounted on boot. Edited within nano /etc/fstab to add above mount command. Have just restarted to make sure its working and a sanity check to make sure that was not the problem. Now updated and has moved to top when doing df -h.

Filesystem      Size  Used Avail Use% Mounted on
udev            3.9G     0  3.9G   0% /dev
tmpfs           787M  1.8M  785M   1% /run
/dev/sda2       110G   12G   92G  12% /
tmpfs           3.9G     0  3.9G   0% /dev/shm
tmpfs           5.0M     0  5.0M   0% /run/lock
tmpfs           3.9G     0  3.9G   0% /sys/fs/cgroup
/dev/loop0       98M   98M     0 100% /snap/core/9289
/dev/loop1       94M   94M     0 100% /snap/core/9066
/dev/sda1       511M  6.1M  505M   2% /boot/efi
/dev/sdb1       1.8T   77M  1.7T   1% /mnt/videofiles

starbasessd, 2020-06-20:
Which linux are you using?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

pearbo, 2020-06-20:
ubuntu server 18.04

starbasessd, 2020-06-20:
sudo chmod -R 666 /mnt/videofiles
It'll request your user password to do root commands.
It should then make a change to the /mnt/videofiles system
you can check by
ls -la .mnt
It should show rw-rw-rw-

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

pearbo, 2020-06-20:
Ah yes must have been working. Usually giving myself root access with sudo -i

shows;

drwxr-xr-x  3 root root 4096 Jun 20 14:40 .
drwxr-xr-x 24 root root 4096 Jun 11 06:32 ..
drw-rw-rw-  3 root root 4096 Jun 20 14:34 videofiles

starbasessd, 2020-06-20:
test to write there, now.
set File Storage, custom path, /mnt/videofiles
click Apply
In Still images set your filename, enable manual snapshots
Click Apply
Click on video feed
click on lower right camera icon
should save the pic in /mnt/videofiles
ls -la -R /mnt/videofiles

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

pearbo, 2020-06-20:
Funnily enough was just trying that with the default path it gives to see if it is saving the photos.

ls -la -R /mnt/videofiles

root@linuxserver:~# ls -la -R /mnt/videofiles
/mnt/videofiles:
total 24
drw-rw-rw- 3 root root  4096 Jun 20 14:34 .
drwxr-xr-x 3 root root  4096 Jun 20 14:40 ..
drw-rw-rw- 2 root root 16384 Jun 20 14:34 lost+found

/mnt/videofiles/lost+found:
total 20
drw-rw-rw- 2 root root 16384 Jun 20 14:34 .
drw-rw-rw- 3 root root  4096 Jun 20 14:34 ..

I can then view the picture browser in motioneye and it shows all of the pictures I have taken. According to the motioneye interface it seems like it is still saving it to the OS drive, showing drive usage as 110gb drive with /mnt/videofiles. Can then delete the photos and the drive usage decreases.

starbasessd, 2020-06-20:
I am at a loss..
I have a VM running Ubuntu 20.04 server
[image: image.png]
as a hub. I walked you through the same setup I use (250GB secondary drive)
[image: image.png]
[image: image.png]

I don't know why you are having issues.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

pearbo, 2020-06-20:
I figured we were getting to that, really appreciate the help though.

Unfortunately unable to see the images posted.

Would running motioneye as an add on within home assistant be causing the issue? 

Every combination I try just seems to only show the main drive. Other than /dev/sbd1/mnt/videofiles is 0,1gb. Thought that might have worked but does seem to only be 100mb in size.

starbasessd, 2020-06-20:
Yes, there could be something in HA causing the issue, over-riding what
you're setting in motionEye. Try checking with them. I tried using HA in
the past, but found it over-kill for security software, and my Alexas
control everything I have.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

pearbo, 2020-06-22:
Little bit of an update for anyone else that may come accross this. I think it might be because HA puts motioneye into a docker container. You have to mount an additional drive into the docker container for motioneye to see it. 

I googled this a bit and decided it was too hard so just used open media vault on a raspberry pi and pointed motioneye to the network share. Tried for ages to samba share within the server but could not get it to work.

Thanks for the help though, learned quite a bit.

starbasessd, 2020-06-22:
Good to know, thanks for the update.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?