motioneye-project/motioneyeos #2232: Remotely Manage Motioneyeos from Outside my Local Network?

joshontech, 2019-12-19:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.
Yes
I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.
Yes
### motionEyeOS Version

I am running motionEyeOS version: motionEyeOS 20190911

### Board Model

I am using the following board/model: Raspberry Pi 3 model B

### Camera

I am using the following type of camera: V4L2 Camera

My camera model is: Logitech C270

### Network Connection

My motionEyeOS unit is connected to the network via: WiFi

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

None

### Log Files

I consider the following log files relevant to this issue: Not Needed

### Question/Issue
I can accsess my camera feed from outside my home network easily that is NOT the issue. I want to be able to accsess my MotionEyeOS settings from outside my network. 

I log into my camera feed in this way. I have Dynamic DNS set up. To connect I simply type my ddns domain name followed by : and than the port I forwarded. Doing that will only get my feed no way to log into my MotionEyeOS at all JUST THE FEED. 

What I would like to do is be able to get to my settings so I can remotely turn the camera on and off so if it is off when I are home and I forget to turn it on when I leave I can remotely do so.

Thanks to anyone who can help! =D

digiltd, 2019-12-31:
you need to forward the right port, it sounds like you are port forwarding directly to the camera stream, try setting up another forward to :80 which is the web interface

joshontech, 2020-02-12:
> you need to forward the right port, it sounds like you are port forwarding directly to the camera stream, try setting up another forward to :80 which is the web interface

Yea that was my issue I was logging into the camera stream directly. I needed to forward the port to the interface not the stream. Thanks for the reply! Sorry my response was so late.

closed 2020-02-12 by joshontech