motioneye-project/motioneyeos #2233: Video uploads aren't being uploaded reliably

sts-ryan-holton, 2019-12-20:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

dev20191220

### Board Model

Raspberry Pi Zero W

### Camera

Official Raspberry Pi V2 NoIR camera

### Network Connection

Wi-Fi, 2.4G

### Peripherals

N/A

### Log Files

I consider the following log files relevant to this issue:

[boot.log](https://github.com/ccrisan/motioneyeos/files/3989234/boot.log)
[messages.log](https://github.com/ccrisan/motioneyeos/files/3989236/messages.log)
[motion.log](https://github.com/ccrisan/motioneyeos/files/3989239/motion.log)
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/3989241/motioneye.log)
 
### The Issue

I've recently installed Motioneye OS, the latest Dev version as I've got connectivity issues with the stable version.

I'm trying to record video continuously in 10 second video clips (I've tried 30 seconds and 60 seconds). I've configured most of the settings, and have got SFTP setup successfully, as indicated when clicking "Test Service", and indeed after applying changes, 1 or 2 video clips will successfully get uploaded with the following file name: `%Y-%m-%d-%H-%M-%S`, however, after these initial uploads, further uploading fails with no visual errors, I check my local FTP server (a Synology, but have also tried Google Drive) and it doesn't upload any more?

reconfiguring some settings, random settings and applying will re-initiate the upload of a few files, new files, and then they stop uploading again.

However, image/photo uploads work without fail.

This sounds like some bug going on with the uploading process. It's the same to Google Drive.

### Screenshots

[image]

sts-ryan-holton, 2020-01-19:
Anyone able to reply?

sts-ryan-holton, 2020-03-05:
/cc @ccrisan

ccrisan, 2020-03-09:
The `motioneye.log` file should contain more details on problems related to uploads. Please provide it.