motioneye-project/motioneyeos #2235: Continuous recording issue dev20191221

cigas4, 2019-12-23:
Hello , 
New pre-release version has an issue with continuous recording. It is related to "Maximum Movie Length" parameter in Movies section. Now it only records a single video of selected length and then stops (no more red frame around video screens). But it should keep recording continuously by selected maximum length (I use 3600 seconds, but it acts the same no matter what movie length is selected). 
20190911 version was working in correct way, but in this new release it stopped working. 

### motionEyeOS Version

I am running motionEyeOS version: dev20191221

### Board Model

I am using the following board/model: Raspberry PI 4B+ 4gb

### Camera

I am using the following type of camera: Network Camera

My camera model is: Reolink RLC-410 and RLC-511

rtmp://IP:1935/bcs/channel0_main.bcs?channel=0&stream=0&user=admin&password=pass

### Network Connection

My motionEyeOS unit is connected to the network via:  Ethernet

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

External Hard Disk

CCA2878, 2020-01-09:
I have the same problem.It makes me annoyed.
So,I want to know if there's any version that doesn't contain this bug.
Please tell me.Thanks.

JohanD78, 2020-01-09:
Been this way for the for the latest dev builds. IIRC there was a Novemebr build that worked but for now i have gone back to the September 11 release.

RPI4 4GB
CSI Camera module
Wifi 5GHZ
256GB SD with 2 day retention

Exact same problem as cigsa4.
I have set to 600 Sec (10 minute segments) and the first is recorded but then it stops. If i set to 1200 sec it records for 20 minutes and stops. One continues recording works but that is not viable...

jasaw, 2020-01-10:
Hi guys, I apologize for not having any spare time to look into this issue. I suspect this is a motion software issue. Can you guys please follow this guide and test motion software in isolation? https://github.com/ccrisan/motioneyeos/wiki/Report-Motion-Issue

If bug still exists, please raise an issue in [Motion Project](https://github.com/Motion-Project/motion/issues).

JohanD78, 2020-01-11:
I have no problem getting the  /data/output/motion.log and the /data/etc/motion.conf but i cant seem to find the  /data/etc/thread-*.conf file

Any help is appreciated :)

Extra info:

I am using my Pi3B for testing, and it worked fine with the October 10 build that was on the SDcard, but the same problem described above showed itself when i upgraded to the december 21 build.

ccrisan, 2020-01-12:
@JohanD78 thread-*.conf has been renamed to camera-*.conf in recent versions.

JohanD78, 2020-01-12:
Thanks. Will do some proper testing tonoght

JohanD78, 2020-01-12:
After running the tests it is still not recording anything after the 600 seconds i have set it to and an FYI:  I have "Motion Detection" turned OFF in settings as i dont see any point having it active when using continuous recording.

I dont know how i forgot to add this in in my first post but i *think* that is some information you would like to have.

As i am just a user and know about what u can expect from one, but is motion a point of failure when i have motion detection turned off?

hgw77, 2020-02-22:
Have the same problem, I reported the issue here https://github.com/ccrisan/motioneye/issues/1653

JohanD78, 2020-02-22:
Actually never tried a USB cam. Doing that now, with csi cam also hooked up and will report back when done testing that.
.
.
.
That was a quick test. Same problem with usb cam. First movie, set to max leanght of 20 seconds, finish as it should but it wont save anything after it.

Cross posting to inform here aswell.

JohanD78, 2020-03-08:
Posting back here aswell: [https://github.com/Motion-Project/motion/issues/1093](url)

"Try to run motion with USB camera only with -d 7 switch, reproduce the issue and upload the fresh log"

[camera-1.zip](https://github.com/ccrisan/motioneyeos/files/4303048/camera-1.zip)

Only usb camera attached with latest 200223 dev