motioneye-project/motioneyeos #2236: Sew MotionOS SD card fails to boot, threatens to ruin an old man's Christmas!

w2tpo, 2019-12-24:
My apologies if this does not belong here. I am an old guy w/ a cognitive issue that makes research a real challenge,,,  and this is such a basic Q that I am hoping that I can get an answer without a lot of run around.

I have a Raspberry PI 4 B with a camera.  I have SD cards that will boot it into Raspbian, Ubuntu and Kali with no problems.

I bought a MotionOS SD card on ebay that said "Raspberry Pi 3".  This will not boot at all.  my monitor never so much as sees a connection.   the Power light is solid red, but the act light flashes once real fast, followed by one slower flash and then goes out.

Hence, I cannot even tell you what version of the OS it is.  but tbe board and camera are brand new.

Seems to me that this is a card fail, and not a "you bought it for the wrong board" issue.

Any guidance would be greatly appreciated.  and Christmas blessings to y'all for taking the time to read thru this.  

My best!
Tom O'Halloran

pius1212, 2019-12-24:
It's most likely a board compatibility issue.  You have said, your motion os card was for RasPi 3.  They have different code inside it.  You bought it for the wrong board litterally

w2tpo, 2019-12-24:
Thank you. I can accept that, it just seemed it to me that any incompatibility would rear its ugly head after it was booted up, and would most likely just affect the camera.

I appreciate you taking the time to get back with me with your input

bsimmo, 2019-12-28:
just download the correct image (you shouldn't need to buy it on a card, spend the money and a decent SD card instead).

instructions start here, on the image download, the Pi4 is at the bottom of the list.

https://github.com/ccrisan/motioneyeos/wiki/Installation