motioneye-project/motioneyeos #2239: MotionEye stops working after 24hs on Raspberry Pi 3 B+ with 3 cameras

mariano-piatti, 2019-12-29:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20190911

### Board Model

I am using the following board/model: Raspberry PI 3B+

### Camera

I am using the following type of camera: 3 Webcams

My camera model is: Microsoft LifeCam HD-3000

### Network Connection

My motionEyeOS unit is connected to the network via: WiFi

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

Micro SD 32GB

### Log Files

I don't have access to log files.

---------------------------------

Hello everyone, I've been using motioneye on my Raspberry Pi 3B+ with 3 cameras for more than a year. In this whole time, motioneye always stopped working after about 24 - 48 hs, showing me this errors when reboot the RPi:
[image]
I bought a new Raspberry Pi 3B+, a new power supply (3.1A) but i still have the same error. I've not tried with a new SD card, I just keep using the same (32GB).
The only way to fix this is by burning the image on SD from scratch and start again.

I've googled and it seems that file system gets corrupted but I don't find the solution.

I will be very grateful with any kind of help.

Ultrabenosaurus, 2019-12-31:
The second line says there's an I/O error mounting the SD card to `/data` and you specifically say you haven't tried a using a new / different SD card. My first instinct is to try a new card as your current one might be damaged or corrupt in some way. 

It seems odd there are lines indicating `post-upgrading to version 20170827: failed` followed by similar errors for 20190127 and 20190307. That said, I use motionEyeOS headless and don't see the boot screen, so maybe `post-upgrading` a version from two years ago is normal behaviour? 

However, back on topic, the `user.notice post-upgrade` lines relate to errors accessing files under `/data` which it obviously can't do due to the second error I already mentioned, where it fails to mount `/data`.

Also note that the late 2018 / early 2019 releases often could not auto-update through the web UI and required manual flashing due to incompatible partition layouts or other filesystem things. If you are not flashing the latest version and then updating through the web UI this could be causing your errors. 

My non-expert recommendation: buy a new SD card and [download 20190911](https://github.com/ccrisan/motioneyeos/releases/tag/20190911) afresh.

mariano-piatti, 2019-12-31:
> The second line says there's an I/O error mounting the SD card to `/data` and you specifically say you haven't tried a using a new / different SD card. My first instinct is to try a new card as your [...]

Hello, thanks for your reply.
I'm buying a new SD card in the next hours, and I'll try it with a fresh 20190911 image. I'm not updating through the web UI so that must not be my problem. I'll update if this worked for me or not.

jasaw, 2020-01-02:
Make sure your SD card storage is not filled to 100% with media files.

mariano-piatti, 2020-01-02:
> Make sure your SD card storage is not filled to 100% with media files.

Thanks for your reply.
I've never reached that amount of space, since I have selected to have the files auto-deleted after 1 day.

I have bought and tried a new SD card (64GB), and I'm having the following issue so far: The MotionEyeOS UI sometimes is not reachable. I can't access the system through web for several hours, then for a couple of minutes I do can and I see that the system has been working normally all the time, but I just can't get in.
Is there any solution?

mariano-piatti, 2020-01-03:
Hello, after changing the SD card for a new one, the problem was solved and I have motioneyeos working normally.

About the problem I mentioned before, I had the HTTP port listening on port 8080 and cameras streaming on ports 8081 - 8083. Turning the camera streamings off and setting the HTTP port on 80 and forwarding to it from 8888 on the router was the solution for me.

Thanks for your help.

closed 2020-01-03 by mariano-piatti

pdisner, 2020-01-04:
I’ve had the same issue. But it isn’t consistent. For example, I have a few cameras that I put together. I’m using cameras that I run straight from the board. I run them all on motioneye. I just set a new one up that is dropping off constantly. I’m just going to try and reinstall and make sure it isn’t something with regards to that. I haven’t tried a usb webcam.   I was curious about how well they work in comparison and probably will buy a nice one in the near future, not only to have another camera, but to for a microphone as well. How well does yours work?
    	
    	

    	Get Outlook for iOS

Closed #2239.

mariano-piatti, 2020-01-05:
Hello, my webcams work really well, I've never tried an IP camera but I will try soon. Although webcams have microphones, there's no way to record sound on motioneyeOS. So, if you're thinking in a webcam, have that in mind.