motioneye-project/motioneyeos #2242: Network cam drops out constantly

Planeguy102, 2020-01-01:
My network camera doesn't seem to want to stay streaming for more than 5 or 10 minutes, before it just stops working. The Raspi stays up, and I can modify settings, which sometimes gets it to work again. Usually, it drops out right after modifying any setting, and it doesn't start working again after I either turn a bunch of options off and then on again, or physically unplug the device and reboot it. Everything else on the Pi works after a drop, it's just the camera that stops working.

### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20190911

### Board Model

I am using the following board/model: Raspberry PI 3B

### Camera

I am using the following type of camera: MMAL

My camera model is: 5MP OV5647 Sensor

### Network Connection

My motionEyeOS unit is connected to the network via: Wifi

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

N/A

### Log Files

I consider the following log files relevant to this issue:

[dmesg.log](https://github.com/ccrisan/motioneyeos/files/4013713/dmesg.log)
[boot.log](https://github.com/ccrisan/motioneyeos/files/4013714/boot.log)
[messages.log](https://github.com/ccrisan/motioneyeos/files/4013715/messages.log)
[motioneye.log](https://github.com/ccrisan/motioneyeos/files/4013716/motioneye.log)
[motion.log](https://github.com/ccrisan/motioneyeos/files/4013717/motion.log)

jasaw, 2020-01-06:
Not sure what the cause of the issue is, but I see the below error messages from your log files.

>Dec 31 16:00:02 meye-ce543dea user.notice throttlewatch: currently: under-voltage throttled 

Get a better power supply.

>2019-12-31 17:55:52: [motioneye]  WARNING: Connect error on fd 15: ECONNREFUSED

>[    9.105724] CIFS VFS: cifs_mount failed w/return code = -5
[    9.141888] CIFS VFS: Unable to select appropriate authentication method!
[    9.141899] CIFS VFS: Send error in SessSetup = -22
[    9.141927] CIFS VFS: cifs_mount failed w/return code = -22
[    9.240315] CIFS VFS: cifs_mount failed w/return code = -5
[    9.315297] CIFS VFS: cifs_mount failed w/return code = -5
[    9.400914] CIFS VFS: cifs_mount failed w/return code = -5
[    9.450478] Status code returned 0xc0000022 STATUS_ACCESS_DENIED
[    9.450518] CIFS VFS: Send error in SessSetup = -13
[    9.450549] CIFS VFS: cifs_mount failed w/return code = -13

Maybe this has something to do with it?

Planeguy102, 2020-01-06:
> Not sure what the cause of the issue is, but I see the below error messages from your log files.
>
>
> Get a better power supply.
>
>
>
> Maybe this has something to do with it?

The under voltage throttling doesn't seem to be the cause, though, because sometimes the camera stays nonoperational after a hard reboot. And, what do those mounting errors mean? I don't have anything but the single partition SD card plugged into the Pi.

limones1319, 2020-04-27:
@Planeguy102 did you find out what causing the under voltage? 

I seem to have the same problem:

Apr 27 09:31:43 cam1 user.notice throttlewatch: currently: under-voltage throttled

And the camera isn't working, but the OS and Pi are booting fine.

I can say that I am using the same power supply and the same cables for at least 1 year. This problem happened a week ago.

limones1319, 2020-04-27:
And I have two cameras with the same configuration and power supply and they both have the same problem.

starbasessd, 2020-04-27:
Have you tried another power brick? They can and do fail. Especially if
borderline maxed out.
You don't state what your setup is.
Which PI
Which Camera
How connected to camera
Network Connection
What power supply and rating, and have you changed it with one not also
failing?
Do you have HDMI monitor connected, with mouse & keyboard, or headless?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

limones1319, 2020-04-27:
I didn't try another power supply, but it's weird that they both failed (if they failed) at the same time, after almost an year of working fine.

They are both RPi2 with 8MP NoIR camera, connected directly to the switch with fast ethernet cables.

I don't remember the power supply rating, and they do not have any mouse, keyboard, or any other dongles attached, just the Pi and the camera.

starbasessd, 2020-04-27:
Well, " Apr 27 09:31:43 cam1 user.notice throttlewatch: currently:
under-voltage throttled" indicated power supply problem. and slowing the
CPU for self protection
Anytime the CPU slows, a lot of networking failures happen.
If you can ssh into the camera try running:
vcgencmd measure_temp
This will report the CPU temp in `C
If it is over 50, not good
If it is over 60, starting to throttle due to temp, indicating other
failures possible
Run command:

vcgencmd measure_volts core

to see what the CPU voltage is, should be 1.3500 volts and stable

vcgencmd get_throttled
(a non-zero result should be investigated).

vcgencmd measure_clock armreports current clock rate in Hz.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

limones1319, 2020-04-27:
These are both of them:

[root@cam1 ~]# vcgencmd measure_temp
temp=47.8'C
[root@cam1 ~]# vcgencmd measure_volts core
volt=1.2000V
[root@cam1 ~]# vcgencmd get_throttled
throttled=0x50005
[root@cam1 ~]# vcgencmd measure_clock armreports
frequency(0)=0

[root@cam2 ~]# vcgencmd measure_temp
temp=49.4'C
[root@cam2 ~]# vcgencmd measure_volts core
volt=1.2000V
[root@cam2 ~]#  vcgencmd get_throttled
throttled=0x50005
[root@cam2 ~]#  vcgencmd measure_clock armreports
frequency(0)=0

starbasessd, 2020-04-27:
vcgencmd measure_clock arm
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
Sorry, missed a space there.
But both are reporting many throttling incidents, both are low voltage.
It's either both Pis are failing, or both power supplies are.
You could try powering one from a PC or Laptop, or one of the USB Power
banks for an additional test.
Make sure it can supply AT LEAST 2.1A (usually shows a double lightning
bolt on the port)
I'd bet something in the power bricks, though.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

limones1319, 2020-04-27:
[root@cam2 ~]#  vcgencmd measure_clock arm
frequency(45)=600062000

[root@cam1 ~]# vcgencmd measure_clock arm
frequency(45)=599998000

starbasessd, 2020-04-27:
At this point:
But both are reporting many throttling incidents, both are low voltage.
It's either both Pis are failing, or both power supplies are.
You could try powering one from a PC or Laptop, or one of the USB Power
banks for an additional test.
Make sure it can supply AT LEAST 2.1A (usually shows a double lightning
bolt on the port)
I'd bet something in the power bricks, though.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

limones1319, 2020-04-27:
I'll try to use a different power supply tomorrow and let you know. The cams are 25km away in a remote location. Thank you.

starbasessd, 2020-04-27:
No problem.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

limones1319, 2020-04-28:
Weird. I changed the power source to a 5V 2.5A and it's the same:

[root@cam1 ~]# vcgencmd measure_temp
temp=50.5'C
[root@cam1 ~]# vcgencmd measure_volts core
volt=1.2000V
[root@cam1 ~]# vcgencmd measure_clock arm
frequency(45)=600000000
[root@cam1 ~]#

This PS was working fine with another Pi with a 4 relay HAT.

limones1319, 2020-04-28:
Still no image.

limones1319, 2020-04-30:
After upgrading one of the cameras to 20190911, the image is back in web UI. I will upgrade the second one as well.

The interesting thing is that both cameras where recording all the time, but there was no image in web UI.

It's weird that both of them experienced the same problem at the same time.