motioneye-project/motioneyeos #2243: motionEye error, no image shown on web

agunggeri, 2020-01-01:
[image]

can I use mobile hotspot to run my rpi? the pi was connected but there's no image shown up on the web.
the led on cam module was turned on, I've tried to resupply the power.
I've install the latest vers of motioneye, and already install the vers 0.41 too, but nothing changed.
is it happen because I use mobile hotspot? can anyone help please?

pdisner, 2020-01-01:
Well  I think it’s because u need a network for the rPI to work thru. But that’s just talking from little knowledge of coding, and this being open sourced, I’ll bet someone here will pick up the challenge

Paul
pdisner@me.com

jasaw, 2020-01-06:
@agunggeri You may have to delete the default cameras and add a new camera. The default cameras are bogus (they point to /dev/video10, 11, 12, which are not camera devices).