motioneye-project/motioneyeos #2247: can't get past login and throttlewatch

rapattack55, 2020-01-04:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: (insert your version here, e.g. 20180314).

### Board Model

I am using the following board/model: (insert your board model here, e.g. Raspberry PI 3B+).

### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

My camera model is: (insert your camera model, e.g. Logitech C290).

### Network Connection

My motionEyeOS unit is connected to the network via: (choose your network connection, e.g. WiFi, Ethernet).

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)

rapattack55, 2020-01-04:
sorry I don't know how to use this site and I am confused with installing motioneyeos on my raspberry Pi 3. so I got up to the login on the pi. this what it says:
welcome to meye-132682b3f
if u want to see user interface point your Web browser to the IP address of this board.
meye-132682b3f login:throttlewatch: currently: under-voltage throttled...
so it repeats the last line again. I am a ubuntu user that's not very proficient but can follow instructions

jasaw, 2020-01-06:
Maybe try powering your Raspberry Pi from a better power supply?

rapattack55, 2020-01-06:
ok yes someone on a forum said I have to get a proper power supply that is 5v 2.5a. hopefully do that today. carla