motioneye-project/motioneyeos #2248: adding netwerk motioneyeos usb camera into motioneyeos not working anymore

alpa2009, 2020-01-05:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: motionEyeOS dev20191221

### Board Model

I am using the following board/model: Raspberry PI 3B

### Camera

I am using the following type of camera: Logitech C290).

### Network Connection

My motionEyeOS unit is connected to the network via:  Ethernet

### Peripherals

I am using the following peripherals that I consider relevant to this issue:
none

In the past I could add a motioneyeos usb based camera into another motioneyeos as a networkcamera (cfr hub setup with central motion/recording).  Now it seems not be possible anymore: or one has to setup a fast networkcamera for the camera itself (which is not possible for USB cams) or a normal setup but still with a MMAL based camera...
Although this is mentioned on the wiki, it still did work for quite some time for me..   I have 9 motioneyeos camera's and 2 nvr's (read: motioneyeos hub setup with central motion/recording) .  Is there any way to set this up again?

ccrisan, 2020-01-25:
We haven't (willingly) changed anything about the possible interconnection scenarios. It should be possible to add a USB-attached camera configured in a motionEye server to another server, using the Remote motionEye Camera type.

If that is not the case, please provide some screenshots indicating the error that you get or the problem that you encounter.

alpa2009, 2020-01-26:
setting up a usb-attached remote motionEye camera (so usb connected to another motioneye camera) doesn't provide the possibility to record on the hub instead of the camera itself...  Furthermore I noticed that some non-motioneye camera's (I only have a few ABUS left) also stopped to record...  and really tried a lot of configs to get it working again (both on latest stable as all later then stable dev versions)  So to solve all of the above, basically I moved all to record individually to SMB share instead of motioneyeos based "NVR".  so using the hub as remote single point of configuration but nothing else...   for the record, running motioneyeos for a couple of years now, and at the moment 11 camera's running on it (on most different possible raspberry pi versions)...

ccrisan, 2020-01-26:
> setting up a usb-attached remote motionEye camera (so usb connected to another motioneye camera) doesn't provide the possibility to record on the hub instead of the camera itself

It never did. That's the way it was designed from the very beginning.

> so using the hub as remote single point of configuration but nothing else

That's what it's supposed to be, so you're probably using it correctly :)

alpa2009, 2020-01-26:
well, only recently started with a dual camera setup (so 1 motioneyos with both CSI and USB camera) where it became obvious...   but had 1 setup with a single USB camera and to the best of my memory this was also recorded on the hub as every single other camera...   don't remember how I configured it, but it was not setup as a fast network camera (as not possible with usb) and not as a remote motioneyeos neither...
But I am not going to discuss with the main developer!  ;o)   keep motioneye(os) alive!  It's the only opensource IP CCTV solution that survived all my research and/or over the years...  ;o)

closed 2020-01-26 by alpa2009