motioneye-project/motioneyeos #2251: Stuck in Bootloop - panic: rebooting in 60 seconds (caused by network)

joshpennington, 2020-01-07:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20190911

### Board Model

I am using the following board/model: Raspberry Pi Zero W

### Camera

I am using the following type of camera: None at the moment

My camera model is: None at the moment

### Network Connection

My motionEyeOS unit is connected to the network via: I have configured WiFi, but boot process does not get that far

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

* Nothing at the moment. Just an HDMI cable and a power cable

### Log Files

I consider the following log files relevant to this issue: Can't get that far in boot process

I am getting stuck in a boot loop on the following step:

* Configuring wired network: no device
panic: rebooting in 60 seconds (caused by network)

The name of my image file is: motioneyeos-raspberrypi-20190911

Am I using the right image for a Raspberry Pi Zero W?

jasaw, 2020-01-10:
Yes you are using the correct image. It's rebooting probably because it doesn't have network connection. Have you tried pre-configuring wifi? Follow the steps here: https://github.com/ccrisan/motioneyeos/wiki/Wifi-Preconfiguration

joshpennington, 2020-01-11:
I actually did have the file put in, but Windows decided to not drop the .txt extension from the file when I created it. 

Problem solved.

closed 2020-01-11 by joshpennington