motioneye-project/motioneyeos #2252: Snapshot URL: with a filename

jasperza, 2020-01-09:
1. I want to use an API to trigger motioneye to take a photo and save it. There is a small icon on the bottom right of the screen which allows the user to trigger taking a snapshot, however I want to do this programatically, what is the URL to do this?
2. I also want to provide the filename , or part of the filename, e.g. a unique number that my program generates, which i can reference to other sensor data later when I download the images , it it possible to provide this?  alternatively, does the snapshot URL return the filename that was created?

Bobo-amg, 2020-01-10:
You can use curl to capture a snapshot with {IP}:{port0}/{camid}/current (eg curl  -o newimg.jpg 10.1.1.253/picture/1/current/) will take a snapshot and save it your local drive.

jasperza, 2020-01-11:
Yes but does that also capture it to the rpi? I am calling it from an IoT device that doesn't have a browser, Local drive or human.

Bobo-amg, 2020-01-12:
Yes it does. It will save the image to the pi that is running the curl command. So I have a pi server that runs the curl command remotely on the 253 camera (also a pi). The image is saved on the server.

I have made this even easier by using Node Red and sending images to Telegram.

jasperza, 2020-01-21:
THanks for your reply, i did call the snapshot URL "http://10.0.0.244/picture/1/current/" and it doesnt create the file on the motioneye device and it doesnt upload it to google drive , when i click the snapshot button it does save it on the motioneye device and it doesnt upload it to google drive. 

These settings work when clicking the "snapshot" button but dont work when calling the snapshot URL
[image]