motioneye-project/motioneyeos #2253: Vstarcam not supported in Motioneye os

digitalxz, 2020-01-10:
Hi 

I have just installed motioneye os on rpi3 and while adding Vstarcam it gives an error saying its not supported. Is there any guide to fix this problem. 

TIA

digitalxz, 2020-01-10:
I have tried adding these streams 
rtsp://192.168.1.101:10554/tcp/av0_0 and http://192.168.1.101:81/snapshot.cgi

and it doesnot works.. 

also why is that the username and password settings in motioneyeos revert to default whenever i refresh the page. like i change the admin and password of vstarcam and when i refresh he page its reset to default. 

[image]
[image]

rmuz8, 2020-01-10:
Try rtsp://admin:password@192.168.1.101:10554/tcp/av0_0

(obviously replacing admin and password with the admin password for the cam) Default I believe was admin:888888

I had a vstarcam up and running last weekend on motioneyeos

azelinc, 2020-02-08:
i tried what rmuz8 suggested, but i cant add the camera as when adding admin:password (replaced it with my actual password), the camera doesnt appear to be able to be selected. so i cant add the camera.
Only rtmp:192.168.1.101:10080/tcp/av0_0 gives me the camera, but it is just a blank black video feed

ahanifen, 2020-04-28:
@azelinc Try entering the URL this way, substituting the UN/PW for your settings:  http://192.168.1.101:10080/videostream.cgi?user=admin&pwd=88888888

fratellino69, 2020-10-04:
Same problem for me....anyone found solution???

azelinc, 2020-10-05:
> Same problem for me....anyone found solution???

i got it to work with this 

rtsp://192.168.0.119:10554/tcp/av0_0

fratellino69, 2020-10-05:
me too. is possible that last version of motioneye has problems in
adding cams?

starbasessd, 2020-10-05:
Setup: RPi0, MotionEyeOS-dev20200907, mmal + Network FosCam rtsp camera,
wifi.
Cannot confirm issue with adding known good Foscam to existing mmal camera
(working) setup in PiZero hub setup.
Same setup works with adding Network Camera (d-Link DCS-5020L) mjpeg stream.
Using same setup from my primary hub (rtsp://192.168.2.15:88/videoMain,
username & password, BOTH are case sensitive) camera registers correctly in
Add Camera, using same video resolution and framerate, get I get the "No
Camera" symbol for about 20-30 seconds, then image appears.
Also work with 20200606.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

gramasx, 2022-02-22:
Finally I got this Vstarcam working in motionEye.

I'm using Vstarcam c7824wip model with motionEye v. 0.42.1, Motion v. 4.3.2  running on Debian 11 within Proxmox container. Procedure likely to be be similar for other older cameras from same vendor running on motionEye.

There's two ways of getting streams from this camera to work on motionEye - either get 360p MJPG stream over HTTP or get any of the RTSP over UDP streams (720p, 480p, 180p). 

1. MJPG HTTP stream.
 
You have to provide link for motionEye using this syntax http://[camera ip]:[port]/videostream.cgi?user=[username]&pwd=[password]

Port above is dynamic and changes every time camera is power-cycled. You have to identify this port for your camera. There's more than one way to do it, I'm using Nmap scanning with all ports scanning option. Nmap scan output looks like this:
Discovered open port 10080/tcp on 192.168.1.150
Discovered open port 56246/tcp on 192.168.1.150
Discovered open port 9600/tcp on 192.168.1.150
Discovered open port 10554/tcp on 192.168.1.150

three ports are fixed - 10080 (for ONVIF), 10554 (for RTSP), 9600. Our http port (in this case) is 56246. 

So in this specific setup scenario video link would look like this: http://192.168.1.150:56246/videostream.cgi?user=admin&pwd=888888

This stream link can be processed by motionEye, however you are getting lower res than that in RTSP streams and you'll have to update port every time camera is rebooted. On the plus - http mjpg stream can be more reliable than udp rtsp.

2. RTSP UDP stream.

You have to tell motionEye that this stream uses UDP and provide link in this syntax (for 720p stream) rtsp://[username]:[password]@[camera ip]:10554/udp/av0_0 

For this in camera settings in motionEye web interface under Video Device settings, Extra Motion Options you should enter following (without quotes): "netcam_use_tcp off". Apply and reboot system or restart motioneye service.

This option sets "netcam_use_tcp off" parameter at the end of camera config file (in my case /etc/motioneye/camera-3.conf ) and these parameters are provided for motion as arguments (or smth like that).

By looking at Motion documentation it seems that Motion expects "rtsp_transport udp" parameter, but this parameter is not implemented in motionEye. Anyway netcam_use_tcp parameter seems to be working.

rtsp stream can provide higher resolution than mjpg one, but can be less reliable. 

good luck,