motioneye-project/motioneyeos #2254: Followed instruction guide. Not detecting WiFi Connection on RPi Zero W

ygreq, 2020-01-11:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: motioneyeos-raspberrypi-20190911.img.

### Board Model

I am using the following board/model: Raspberry Pi Zero W.

### Camera

I am using the following type of camera: Raspberry Pi 3 Night Vision Fisheye Camera 5MP OV5647 130 Degree Focal Adjustable Camera.

My camera model is:

### Network Connection

My motionEyeOS unit is connected to the network via: WiFi.

### Peripherals

I am using the following peripherals that I consider relevant to this issue: NA

### Log Files

I consider the following log files relevant to this issue: Configuring wire network: no device

I will give all details here as they appeared. I used Windows to create the Bootup disk for MotionEyeOS on a 16GB SD card from Kingston. I formated the drive using the recommended app, I then used Etcher to put the motionEyeOS to the SD card. 2 partitions were created, one that contained some files like.. and another one that windows asked me to format. Of course I did not. BTW, once it finished installing with Etcher, I had to take the SD card out and put it back in in order for the SD card partitions to be seen by Windows. I then created a txt file that I renamed wpa_supplicant.conf. I checked that there was no txt file extension. (I activated the necessary view option in Windows in order to see file extensions for known file types.) I put the following code into the file:

country=RO
update_config=1
ctrl_interface=/var/run/wpa_supplicant
network={
  ssid=“MYSSID”
  psk=“MYPASSORD”
}

I tried other variations: RO written with small letters (RO is my country code btw), I added   id_str="extn2" at the end after the psk line as I've seen in a tutorial. One time I put US as country code. 

I put the SD cand into the Raspberry Pi Zero W and once it started booting, an error appeared that said 
"Configuring wire network: no device". My guess is that it did not detect the wifi configuration from the wpa_supplicant.conf file. 

I did this for about 9 times, every time starting from the beginning with formating the disk. I checked https://github.com/ccrisan/motioneyeos/wiki/Troubleshooting#wireless-connection-problems and https://github.com/ccrisan/motioneyeos/wiki/Wifi-Preconfiguration and made no mistakes from my point of view. 

I don't know what else to do. Please help!
ygreq

starbuck93, 2020-01-11:
[image]

Same issue here

starbuck93, 2020-01-11:
Fixed my issue by using an USB-Ethernet connector, plugging into a monitor and getting the IP address, and SSHing into the Pi Zero W, making sure `/data/etc/wpa_configuration.conf` was correct and rebooting.

ygreq, 2020-01-11:
Thank you for your input, starbuck93! May I ask that you share a link of the product you are using? 
Where is the /data/etc/ folder? Is it on the second partition that cannot be accessed from Windows? When I copy/paste the wpa_configuration.conf, I know I put it in the root folder. And there are no /data/etc/ folders present. And can you please elaborate? 

Thank you so much!

starbuck93, 2020-01-11:
Sure! It's just a [TP-Link Ethernet to USB adapter I got on Amazon](https://smile.amazon.com/gp/product/B00YUU3KC6). And you need a micro USB to USB adapter like [these ](https://smile.amazon.com/dp/B01K5SXTNG/)for example.

the `/data` mount is only accessible when you log into the linux system. Once you get the IP address, either from plugging in a monitor or scanning your network then it'll be pretty easy to set up wifi. You can edit `wpa_supplicant.conf` with SSH or a program like WinSCP. I think it only moves the `wpa_supplicant.conf` file from `/boot/` (the one you can see on Windows) the first time you power on the Pi Zero with a fresh SD card image.

ygreq, 2020-01-13:
Hi starbuck93! I managed to follow your footsteps. I am stuck at getting into the raspberry via ssh. It's asking for a password and I tried "raspberry" (without the quotes as a password. That did not work. Also tried the password I input in MotionEye. Didn't work either. Any suggestions?

Will try to replicate your steps as well, andrewerrington.

ygreq, 2020-01-13:
Considering I managed to install MEye with the help of your advice (USB to Ethernet cable) once I got in, I changed the settings in Network. I activated WiFi and input there the credentials. 

Everything seems to work. I still would like to know what the password for raspberry in ssh is though. :)

Thank you so much!

starbuck93, 2020-01-14:
@ygreq great! The ssh username is root and there is no default password as far as I know. If you plug into a monitor, to log in, just type in `root` as the username and don't type anything, just hit enter for the password.

andrewerrington, 2020-01-14:
And suddenly it all works. I re-flashed the 20190911 image with Etcher and copied the working wpa_supplicant.conf from the other Pi which was working. I moved the Pi Zero W closer to the WiFi Access Point, in case the signal was weak. It booted!

I moved it back to the bench, where it hadn't been working, and expected it to fail. It booted!

I installed it in its intended location where I first noticed the failure. It booted!

I have used the same SD card for these experiments. Sometimes I installed MotionEyeOS, sometimes I installed Raspbian Buster. Finally, I did nothing different, but it works. That's the kind of error I hate, but, it's working, just like the other one, and just as I had expected and hoped for.

Thanks! I really like MotionEyeOS!

ygreq, 2020-01-15:
Hi, andrew! Can you paste the code you wrote on the wpa_supplicant.conf file, please? Like I mentioned before, I found many examples. 

Be careful to change the SSID and password before copying :D  

Thank you, starbuck93! I will try that for sure!

andrewerrington, 2020-01-19:
Hi there,

I copied mine from the Raspberry Pi web site. It looks like this (this is the actual file, edited of course):
```
ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country=NZ

network={
    ssid="mynetwork"
    psk="mypassword"
}
```
It was automatically copied to _/data/etc/wpa_supplicant.conf_

Change 'NZ' to your country code.

ygreq, 2020-01-21:
Thank you so much, Andrew and starbuck93! I will check everything again tomorrow and close this

maltaresk, 2020-02-22:
i m also facing the same problem with wp_supplicant file. trying repeatedly the same sequence. hoping some day it will boot.
we all are student of science and engg. there is no pinpointed error and proved method. only luck and chance is playing game here.
i m feeling happy that some people have been able to boot it.
some are still trying.
i m open to all kind of suggestions and opinions ... lets share .... here ...

maltaresk, 2020-02-22:
> Fixed my issue by using an USB-Ethernet connector, plugging into a monitor and getting the IP address, and SSHing into the Pi Zero W, making sure `/data/etc/wpa_configuration.conf` was correct and rebooting.

dear starbuck pl let me understand you answer , can you explain. thanks

maltaresk, 2020-02-22:
i think such repeated nature of problems can be shortout by motioneye easly .
m i wrong ?

starbasessd, 2020-02-22:
What are you trying to do?
Connecting to a WiFi using wpa_supplicant.conf file?
Is it a home or corporate network?
What version of Pi? PiZeroW should use the on board wireless without issue.
Pi3B, Pi3B+, and Pi4B will also work without issue.
PiZero (no wireless) and Pi1 or Pi2 will work if you use an EDIMax USB wifi
dongle. Others work too, but EDIMax is my favorite.
You might be able to get an USB Ethernet  dongle to work also.
All versions of motionEyeOS on Pi require some sort of network connection,
or tell it that you want it to stand alone without networking.
I have posted the directions for the stand alone setup here, in other
responses.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

bbillp, 2020-02-22:
I had better luck loading an old version of Motioneye to setup WiFi then clicked the UPGRADE button for the newest version.        Bill ...

maltaresk, 2020-02-23:
> Fixed my issue by using an USB-Ethernet connector, plugging into a monitor and getting the IP address, and SSHing into the Pi Zero W, making sure `/data/etc/wpa_configuration.conf` was correct and rebooting.

evenif i get ip address by ethernet connection. however ultimately the motioneye is required to boot completely, to work on it. correct ?
or after getting ip address of pi zero w , it will act normal and it will boot and camera will be working?
pl share your experince.

starbasessd, 2020-02-23:
Unless you set OS_NETWORKLESS="true" in os.conf, as soon as you unplug
Ethernet it will panic and go into a reboot loop.
If you want the PiZeroW to work in a stand alone mode, you should follow
the attached instructions.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

rmboyd, 2020-03-14:
I bet you created the file in notepad rather than Notepad++
I made this mistake. pulled my hair out for hours. Fixed for me

towfeeqf, 2021-02-01:
I was able to resolve a similar issue.

It has to do with the file extension. You may check the link below.

https://github.com/ccrisan/motioneyeos/issues/2537#issuecomment-770477111

meyeipuser, 2021-11-14:
I have problem to connect via wifi, motioneyeos, connection is done, but no images seen. (even i have my meye-xxxxxxxx ip), but i can connect via ethernet cable RJ45, it works perfectly. Please, someone, help me by sending me information to "qd.loui@free.fr".
My raspberrypi model raspberrypi B. Thanks.

starbasessd, 2021-11-14:
@meyeipuser 
There isn't wifi on Pi B, unless you are plugging in a WiFi dongle. 
Which version of motionEyeOS are you using?
When connected to the Pi via ethernet, have you set the wifi information up?
Settings, Network
Did you copy a wpa_supplicant.conf file to the boot partition right after imaging the SDCard, before putting it into the Pi?
Are you connecting to http://ip_address ? Or something else?

meyeipuser, 2021-11-16:
Hin sir :
   Thanks for your reply. Here what i have done :

   Yes, i have plugged in a wifi dongle.

   The version used : motioneyeos-raspberrypi-20200606

   I have configured Network of motioneyeos setting
   I have downloaded the wpa_supplicant.conf into the root
   	Ctrl_interface=DIR=/var/run/wpa_supplicant
	GROUP=netdev
	Update_config=1
	Country FR
	Network{
	   Ssid= « xxxxxxx »
	   Psk= « xxxxxx » 
	   Key_mgnt=WPA-PSK
	}
	I can connect via wifi, to meye-xxxxxxxx ip, browser shows meye-xxxxxxxx, but i have not images on the screen.

	If i connect to motioneyeos with Ethernet cable RJ45, it works perfectly. But not with wifi.
	
	Please help me in this issue.

Envoyé à partir de Courrier pour Windows

De : starbasessd
Envoyé le :dimanche 14 novembre 2021 16:22
À : ccrisan/motioneyeos
Cc : meyeipuser; Mention
Objet :Re: [ccrisan/motioneyeos] Followed instruction guide. Not detectingWiFi Connection on RPi Zero W (#2254)

@meyeipuser
There isn't wifi on Pi B, unless you are plugging in a WiFi dongle.
Which version of motionEyeOS are you using?
When connected to the Pi via ethernet, have you set the wifi information up?
Settings, Network
Did you copy a wpa_supplicant.conf file to the boot partition right after imaging the SDCard, before putting it into the Pi?
Are you connecting to http://ip_address ? Or something else?

starbasessd, 2021-11-16:
@meyeipuser 
Try dev20201026. There are several issues with 20200606 that are resolved in dev20201026.
Can you take a screenshot of what you see when you try to connect to http://ip_address of the Pi?

meyeipuser, 2021-11-17:
Dear sir :

	I can’t no longer take a Screenshot of  screen, because i have cleaned my sd card. Anyway, i can describe the screen : in the upside left corner, in a pane browser shows « meye-xxxxxxxx », but no ip,
The screen is « white », no image displayed, i know that i am connect to motioneyeos, just, i have not images of camera.

	I will try dev20201026, i will inform you, the outcome.
		Thanks

Envoyé à partir de Courrier pour Windows

De : starbasessd
Envoyé le :mardi 16 novembre 2021 14:02
À : ccrisan/motioneyeos
Cc : meyeipuser; Mention
Objet :Re: [ccrisan/motioneyeos] Followed instruction guide. Not detectingWiFi Connection on RPi Zero W (#2254)

@meyeipuser
Try dev20201026. There are several issues with 20200606 that are resolved in dev20201026.
Can you take a screenshot of what you see when you try to connect to http://ip_address of the Pi?

closed 2022-01-19 by starbasessd