motioneye-project/motioneyeos #2256: How to access HTTPS cameras?

Shaker1978, 2020-01-11:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20190911

### Board Model

Raspberry PI 3B+

### Camera

Some internet HTTPS webcam

### Network Connection

My motionEyeOS unit is connected to the network via: WiFi

### Peripherals

### Log Files

I consider the following log files relevant to this issue:
Line from motioneye.log:
ERROR: mjpg client timed out receiving data for camera 4 on port 8084`
Line from messages.log:
Invalid netcam service '????' - must be http, ftp, mjpg, mjpeg, v4l2 or jpeg.

In the view it says: "Unable to open video device".

I wonder if nobody else is trying to accessing HTTPS cameras. Is there anything that can be done to make MotionEyeOS access https urls? I will soon have HTTPS local cameras. Will I not be able to integrate them into MotionEye?

ccrisan, 2020-01-25:
I don't think Motion supports HTTPS network cameras: https://motion-project.github.io/motion_config.html#netcam_url. If it does, I'll be happy to update motionEye to allow such an URL.

Shaker1978, 2020-01-25:
You're right. I cloned the motion project and added some OpenSSL code to handle https-addresses without having any experience with C++ development on linux. Maybe you can help me how to include the OpenSSL libraries successfully for all the autoconf stuff. When compiling, it just don't find the OpenSSL classes I use. How do I include OpenSSL correctly in such a project?

ccrisan, 2020-01-25:
That sounds more like a task for [Motion](https://github.com/Motion-Project/motion) than a motionEye task.

Shaker1978, 2020-01-25:
Ah, for sure, thanks for replying. I keep this thread updated, if my mod works. At the moment it's just a small experiment, due to my metioned lack of experience in this project.

bldewit, 2025-01-16:
I am struggling with the same issue as I have several HTTP-based IP-cameras.
On the Github page for Motion it says:

The applications include the ablity to:
Create videos or save pictures of the activity
Passthrough recording from many IP cameras
View live stream of cameras
Invoke scripts when activities occur
Log activity into multiple types of databases
Fully customizable masks for privacy or motion detection
**_Full tls(https) support_** with authentication for webcontrol and streams
Use with many types of devices
Network cameras via RTSP, RTMP and **_HTTP_**
PI cameras
V4L2 webcams
Video capture cards
Existing movie files

Has this  since been deleted (or added) or is there another reason why HTTP is not supported by MotionEye?