motioneye-project/motioneyeos #2264: Using Access Point but SSH not working

derekbez, 2020-01-16:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS 20190911

I am running motionEyeOS version: (insert your version here, e.g. 20180314).

### Raspberry Pi 3A+

I am using the following board/model: (insert your board model here, e.g. Raspberry PI 3B+).

### Camera

I am using the following type of camera: MMAL Camera

My camera model is: camera not connected yet - still in initial setup

### Network Connection

My motionEyeOS unit is connected to the network via: WIFI, using Access Point, as per https://github.com/ccrisan/motioneyeos/wiki/WiFi-Access-Point

### Peripherals

I am using the following peripherals that I consider relevant to this issue: None

### Log Files

I consider the following log files relevant to this issue:  
[motioneye (1).log](https://github.com/ccrisan/motioneyeos/files/4072973/motioneye.1.log)
[boot (1).log](https://github.com/ccrisan/motioneyeos/files/4072975/boot.1.log)

Background / context:
My intention is to create a MotionEyeOS camera that triggers a DSLR out in the field.  The separate circuit to do this triggering in known to work.  I am using Action Buttons to trigger a GPIO pin, as per https://github.com/ccrisan/motioneyeos/wiki/Action-Buttons

Now I want to use the Access Point functionality because there is no wifi out in the field.

Access Point works with the given instructions.  ie: I formatted the SD card, flashed it, copied into the Boot partition:  hostapd.conf, dnsmasq.conf and os.conf.  MotionEye boots up, and I can connect to the Access Point, and I can use a browser to administer things as normal.  

But I still need to get the action-button files onto the device.  Like : /data/etc/light_on_1 and /data/etc/light_off_1, as mentioned in the Action-Buttons wiki page.

From the browser, under Services, SSH Server is ON.

From my PC, once I have connected to the Access Point (and have browser running the ME admin screen), I open PuTTY and point it to the device.  A few seconds later it comes back with a fatal error : Connection Refused.

Are there additional steps to use SSH with the Access Point?
Or
Is there a different way I can get the files onto the sd card (and make them executable)?
Is there a way to toggle between Access Point and Wifi?

mhaserodt, 2020-01-23:
Same issue here using Raspberry PI Zero W.  I've also been watching a thread where others reported the same issue in October.

https://github.com/ccrisan/motioneyeos/issues/2179

unstatusthequo, 2020-06-30:
Same issue with SSH "connection refused" when attempting to SSH while connected to RPi as AP. Though I'm on the 202002 version.

starbasessd, 2020-06-30:
I think it was supposed to be fixed in the 20200606 version, but may not be
working.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

unstatusthequo, 2020-06-30:
Yeah, I had the 20200606 version, but had to switch back to 202002... because on the most current version, I can't even get the AP to start as a standalone device. 
[https://github.com/ccrisan/motioneyeos/issues/2428#issuecomment-650494822](url)

Mohamudo, 2022-07-02:
Was there ever a fix for this? I am having the same issue was well.

If i use 20200606 it looks like hostapd stops working because of ssh and the access point stops showing up.

if i use 20200203 the access point works, but sshd is down.

mhaserodt, 2022-07-02:
> Was there ever a fix for this? I am having the same issue was well.
> 
> If i use 20200606 it looks like hostapd stops working because of ssh and the access point stops showing up.
> 
> if i use 20200203 the access point works, but sshd is down.

The fix for me was to make sure to use ap0 instead of wlan in the config files.  Newer versions of MEOS require a slightly different setup.

Check this out and see if it helps any.

https://github.com/motioneye-project/motioneyeos/wiki/WiFi-Access-Point

Mohamudo, 2022-07-03:
> The fix for me was to make sure to use ap0 instead of wlan in the config files. Newer versions of MEOS require a slightly different setup.
>
> Check this out and see if it helps any. [...]

Thank you that worked