motioneye-project/motioneyeos #2265: Raspberry Pi 4 - Overlock 2147

bobster316, 2020-01-17:
Is it possible to overclock the raspberry pi 4 4gb to 2147, i currently run at 2000 without any problems, if i increase to 2147 the pi fails to boot.

I am trying the following parameters in the config file

gpu_freq=750
arm_freq=2147
over_voltage=6

If i reduce the arm_freq=2000 the system boots up okay.

bobster316, 2020-01-29:
Any updates please...

bobster316, 2020-02-17:
Ive seen a few videos of people upgrading their Pi 4's to overlock to 2147 using the latest firmware.
Are there any plans to update motioneyeos ?

ccrisan, 2020-02-17:
Yes! Let's have the current prerelease validated and turned into a stable release. Then we'll do an upgrade of everything, from BuildRoot to all kernel & firmware versions.

starbasessd, 2020-02-17:
Would there be necessarily a performance boost on the Pi4?

Another q:  Can we get a "Manual" time / date option for not connected PIs?
In addition to HTTP, NTP, and SNTP?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

jasaw, 2020-02-18:
@starbasessd 
>Another q:  Can we get a "Manual" time / date option for not connected PIs?
In addition to HTTP, NTP, and SNTP?

You could create a new issue here and we can have the discussion in the thread.

This is quite easy to implement, but I question how useful it is. A button that triggers `ntpdate` might be more useful?

starbasessd, 2020-02-18:
For stand alone - non internet connected motionEyeOS RPi's. I've helped
several users create the
https://github.com/ccrisan/motioneyeos/wiki/WiFi-Access-Point

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-02-18:
Done.  #2297

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?