motioneye-project/motioneyeos #2269: Raspberrypi zero w gets errors and reboots

Zach0331, 2020-01-28:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: motioneyeos-raspberrypi-20190911.img.

### Board Model

I am using the following board/model: Raspberrypi zero w.

### Camera

I am using the following type of camera: Raspberrypi zero w camera.

My camera model is: Raspberry pi camera v2.

### Network Connection

My motionEyeOS unit is connected to the network via: WiFi.

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)
 
[dmesg.log](https://github.com/ccrisan/motioneyeos/files/4120347/dmesg.log)
[messages.zip](https://github.com/ccrisan/motioneyeos/files/4120349/messages.zip)

MotioneyeOs keeps getting errors and then rebooting after awhile.

ccrisan, 2020-02-02:
There's a huge number of stack traces in your `dmesg.log`. I'd suspect a hardware issue (including possible overheating).

saivert, 2020-02-07:
Or you could remove Raspberry Pi 1/Zero from the supported devices and get far less bug reports like this. It is too weak hardware. It gets the same treatment by almost all the open source projects out there. Nobody is interested in investing serious time to optimize for such weak hardware. The CPU is the same as used in the first Roku device and worked fine as a simple Netflix set top box SoC with properly optimized OS (with licensed VideoCore stuff I bet).
Nobody really cares about the Zero or firs gen Pi for anything but simple GPIO/sensor stuff. I'm sure the Raspberry Pi 2 is also going to be soon forgotten.

msiqx9, 2020-02-24:
> Nobody really cares about the Zero or firs gen Pi for anything but simple GPIO/sensor stuff. I'm sure the Raspberry Pi 2 is also going to be soon forgotten.

I have several Zero W devices running MotioneysOS...they are fine.