motioneye-project/motioneyeos #2270: Using MotionEyeOS and Importing Python Modules, and follow up question.

DdraigX, 2020-01-29:
Howdy,

I'm currently running motionEyeOS 20190911. I am trying to set up a GPIO pin to turn off and on the ir cut filter on my NOIR camera based off the sunrise/sunset period.  I have managed to create a quick python script to do that. However I'm trying to import suntime module https://pypi.org/project/suntime/) Is there a way I can do this with the python on the system currently? Pip doesn't seem to work.

Follow up question here: I installed this before realizing that it used Buildroot so I'm not familiar with it.  In order to get this to work I'm willing to install regular raspian and then motioneye, but how would I make the file system read-only etc as I like that feature of motioneyeos. How would I go about doing that instead? 

Thanks in advance for a great tool.

HOD42, 2020-03-09:
For the filesystem changes:

mount -o remount, ro /
mount -o remount, ro /boot

To reverse, do:

mount -o remount, rw /
mount -o remount, rw /boot

HOD42, 2020-03-09:
Out of interest which camera is that? I didn't know you could enable/disable the IR filter!

starbasessd, 2020-03-09:
You can also turn on Debugging Mode to enable Read/Write, and turn Debug
Mode back off to turn off Read/Write mode.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

DdraigX, 2020-06-07:
> Out of interest which camera is that? I didn't know you could enable/disable the IR filter!

Sorry for the late reply, wasn't getting email alerts on this. I was using  the NOIR camera, and it has a GPIO pin that you can enable or disable the IR.

DdraigX, 2020-06-07:
> For the filesystem changes:
> 
> mount -o remount, ro /
> mount -o remount, ro /boot
> 
> To reverse, do:
> 
> mount -o remount, rw /
> mount -o remount, rw /boot

I was able to get read/write stuff setup, but because it's a striped down version I can't fully install pip on the device. I ended up loading it onto a raspian image to control.

starbasessd, 2020-06-07:
It's easier (although takes 2 reboots)
Debug ON = both file systems R/W
Debug OFF = both file systems R/O

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

HOD42, 2020-06-07:
> Sorry for the late reply, wasn't getting email alerts on this. I was using the NOIR camera, and it has a GPIO pin that you can enable or disable the IR.

Ah! So it's just got attached IR LEDs. I thought you meant you could disable/enable the IR filter on the camera (which would be interesting).

DdraigX, 2020-06-07:
> > Sorry for the late reply, wasn't getting email alerts on this. I was using the NOIR camera, and it has a GPIO pin that you can enable or disable the IR.
> 
> Ah! So it's just got attached IR LEDs. I thought you meant you could disable/enable the IR filter on the camera (which would be interesting).

You can, it's called an IR Cut filter. Have to solder a pin on it and then send a gpio high or low command. I was hoping initially that I could coordinate it to turn on or off depending on sunrise or sunset. However, that was an older model that I shorted out and broke and most of them I believe have an photo-resistor to  trigger the filter on or off if there is enough light.   What I believe I discovered is that python on the motioneyeos image is pretty stripped down so I wasn't able to install some python modules that I wanted to use.  I got the IR Cut filter working though, but couldn't coordinate sunrise/sunset.

https://www.waveshare.com/wiki/RPi_IR-CUT_Camera

HOD42, 2020-06-08:
> https://www.waveshare.com/wiki/RPi_IR-CUT_Camera

COOL! I'm going to have to order one of those! I didn't even know that was "a thing". Thanks for sharing.