motioneye-project/motioneyeos #2274: OS does not boot if the default gateway is blocking

slackwaree, 2020-02-02:
Motion Eye OS latest GIT for Raspberry PI 3.

If the set default gateway is blocking internet traffic or unreachable it seems that MotionEyeOS is in either a stuck or crashed state because it never gets to the point to start any services.

The device is still pingable but since no services reachable it is no longer possible to login on web or ssh in this case. 

This should NOT be the desired behavior. For private camera systems no internet access should be ever required.

ccrisan, 2020-02-02:
motionEyeOS needs real date/time. If you have a local NTP server or any other means to provide that, just configure it in [date.conf](https://github.com/ccrisan/thingos/wiki/date.conf) and you won't need Internet access anymore.

slackwaree, 2020-02-04:
Ok it just takes a lot longer. About 5-10 minutes to come back.
If the default gateway is working the pi comes back in less than a minute. 
Easy to test just create an iptables FORWARD rule on the gateway which blocks all other traffic.
NTP is not relevant for me as it is another machine on the same LAN for me without any firewall rules to block that.

It seems it still tries to connect to something before it starts the services:
sh: connect to host 10.1.0.5 port 22: Connection refused

closed 2020-02-04 by slackwaree