motioneye-project/motioneyeos #2275: My r8188eu USB wifi module doesn't work

denis4net, 2020-02-02:
It seems that r8188eu is not included into the distro.

### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: motioneyeos-orangepione-20190911

### Board Model

I am using the following board/model: Orange Pi PC One

### Camera

I am using the following type of camera: V4L2

### Network Connection

My motionEyeOS unit is connected to the network via: ethernet

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

USB WiFi adapter based on Relatek 8188.
`Bus 003 Device 002: ID 0bda:8179`
Linux module is `r8188eu`

### Log Files

I consider the following log files relevant to this issue:

```
[root@meye-83c5eeef 5.1.9]# dmesg  | tail -n 10
[    2.517600] EXT4-fs (mmcblk0p3): mounted filesystem with ordered data mode. Opts: (null)
[    2.894896] udevd[210]: starting version 3.2.7
[    2.940346] udevd[211]: starting eudev-3.2.7
[    9.064104] Generic PHY 0.1:01: attached PHY driver [Generic PHY] (mii_bus:phy_addr=0.1:01, irq=POLL)
[    9.066458] dwmac-sun8i 1c30000.ethernet eth0: No Safety Features support found
[    9.066477] dwmac-sun8i 1c30000.ethernet eth0: No MAC Management Counters available
[    9.066487] dwmac-sun8i 1c30000.ethernet eth0: PTP not supported by HW
[    9.068101] dwmac-sun8i 1c30000.ethernet eth0: Link is Up - 100Mbps/Full - flow control rx/tx
[   11.320850] system date is 2020-02-02 14:42:40
[  305.607143] usb 3-1: new high-speed USB device number 2 using ehci-platform
[root@meye-83c5eeef 5.1.9]# ip l
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq qlen 1000
    link/ether 02:81:83:c5:ee:ef brd ff:ff:ff:ff:ff:ff
[root@meye-83c5eeef 5.1.9]# lsusb
Bus 005 Device 001: ID 1d6b:0002
Bus 003 Device 001: ID 1d6b:0002
Bus 001 Device 001: ID 1d6b:0002
Bus 004 Device 001: ID 1d6b:0001
Bus 002 Device 001: ID 1d6b:0001
Bus 003 Device 002: ID 0bda:8179
```

monreal, 2020-04-02:
I have three wifi devices with the same chipset. Unfortunately brcmfmac (build-in RPI 3 wifi) does not work either. It would be nice to have some idea what kind of wifi keys work, because most I can find seem to be realtek ones.

starbasessd, 2020-04-02:
If it works in a current Raspbian image, it should work in motionEyeOS.
I have a bunch of different ones
rtl8192cu Edimax
rtl8712u Asus N10
BCM20702A0 Plugable (BT & WiFi)
rtl8192cu Xerox Wireles N
rt2x00 RALink Long range
rtl8192cu LB Link long range
All work with both Raspbian and motionEyeOS.
Does your OTG cable or hub work with other devices?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

monreal, 2020-04-02:
Got my r8188eu working by ssh'ing in when connected via cable and doing the following:

iwconfig wlan0 mode Managed
iwconfig wlan0 essid myssid
service wifi restart
dhclient wlan0

Now the wifi even connects fine after a reboot. Maybe there is something that needs to be added to the setup or init scripts?

monreal, 2020-04-02:
Spoke too soon, now it fails again....

monreal, 2020-04-02:
Problem solved: SSIDs are case-sensitive after all :-)

starbasessd, 2020-04-02:
Yep. You can sometimes get away with it in Windows, but not in Linux or
MacOS...
If you set up your wpa_supplicant.conf file ahead of time, and copy to the
/boot partition when imaging, you shouldn't have any issues.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?