motioneye-project/motioneyeos #2278: No boot and rootfs partition

ruediheimlicher, 2020-02-04:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version 4

I am running motionEyeOS version: motioneyeos-raspberrypi2-20190119

### Board Model

PI 3B+

### Camera

I am using the following type of camera: raspi camera  V4L2.

My camera model is: (insert your camera model, e.g. Logitech C290).

### Network Connection

My motionEyeOS unit is connected to the network via: Ethernet.

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)

ccrisan, 2020-02-04:
You can't possibly ask us to help you with your problem when all you provide is a title and a version.

ruediheimlicher, 2020-02-04:
Hello
I am trying to write motioeyos to a sd card.
MacBook Pro macOS 10.14.6
I am working with Raspberry Pi since many years and never had a similar problem:
I wrote to the card with

`sudo` bash  ./writeimage.sh -d /dev/disk4 -i /Users/myname/Desktop/motioneyeos-raspberrypi2-20190119.img `

resulting in 

`310+0` records in
310+0 records out
325058560 bytes transferred in 93.942505 secs (3460186 bytes/sec)
 * mounting sdcard
Unmount of all volumes on disk4 was successful
 * unmounting sdcard
 * you can now remove the sdcard`

On the card there are 2 partitions with the right contents, but not with names boot and rootfs, but BOOT, Untitled. And teh pi does not start.
The card was formatted with ext4, but I gut the same results with FAT and ExFat.
I also tried:
on Mac:
downloading motioneyeos-raspberrypi2-20190911.img.xz and expanding with xz in terminal
writing to card with ApplePiBaker and balenaEtcher
Same with motioneyeos-raspberrypi2-20190119.img.xz.

on Windows:
downloading expanding motioneyeos-raspberrypi2-20190911.img.xz and expanding with the Unarchiever
writing to card with  balenaEtcher
Same with motioneyeos-raspberrypi2-20190119.img.xz.

Cards formattet with FAT or exFAT
Always with the same result.

I used a 8GB and a 16GB card witch both were used before and after  with a normal raspberry stretch image without problems.

Where is my mistake?

Screenshots:

Partition Untitled
[image]

Partition NO NAME
[image]

ccrisan, 2020-02-04:
> On the card there are 2 partitions with the right contents, but not with names boot and rootfs, but BOOT, Untitled.

The partition name (aka label) does not really matter.

> The card was formatted with ext4, but I gut the same results with FAT and ExFat.

The formatting and partition layout prior to writing an OS image is irrelevant. The image always comes with its partition table and partitions themselves.

What I see in your screenshots looks fine for a motionEyeOS SD card.

Can you please make a photo of your Raspberry Pi board and attach it here?

ruediheimlicher, 2020-02-04:
Thanks for the response
photo as attachment.

The same board RP3+ with the same connections to keyboard and Ethernet and  on the same table worked today with the same card running eg. my homeautomation task, other blank stretch and buster image. I run an installation of motioneye about three years ago wich I destroyed with updating to the newest version of motioneye.
I also tried on a RP2+.
Gruss
Ruedi

ccrisan, 2020-02-04:
Your attachment did not get through. Replying via email won't keep your attachments. You need to use GitHub's website.

> The same board RP3+...

If it's a Raspberry Pi 3, you need to use the OS image corresponding to, well, Raspberry Pi 3. From what you've been describing above, you've been trying it with the Raspberry Pi 2 OS image variant.

closed 2020-02-04 by ccrisan

ruediheimlicher, 2020-02-05:
HAllo Calin
success: changing the power source to a better one and raspi starts. Not the first one with this issue, I think.
Thanks for your help.
Gruss
Ruedi