motioneye-project/motioneyeos #2280: cannot add my cameras

holzfelix, 2020-02-06:
Hi I use the current motion eye os on a rasperry 3b+

When I try to add my camera like this as network camera:

rtsp://admin:123456/192.168.178.100/stream0/

I goth the error Name or service not known or camera not supported. ... 

its a simple rtsp stream

ccrisan, 2020-02-06:
`rtsp://admin:123456/192.168.178.100/stream0/` does not look like a valid RTSP URL. Try:
 * url = rtsp://192.168.178.100/stream0
 * username = admin
 * password = 123456

holzfelix, 2020-02-07:
works thanks

closed 2020-02-07 by holzfelix

joeydream, 2020-07-20:
how about there is only one row for you and you must put the admin and password in one line?

> `rtsp://admin:123456/192.168.178.100/stream0/` does not look like a valid RTSP URL. Try:
> 
> * url = rtsp://192.168.178.100/stream0
> * username = admin
> * password = 123456