motioneye-project/motioneyeos #2282: SSHD daemon disabled for stand alone setup

starbasessd, 2020-02-07:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.  Yes

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue. Yes

### motionEyeOS Version 

I am running motionEyeOS version: (20200203 & 0911).

### Board Model

I am using the following board/model: (Raspberry Pi ZeroW and Raspberry Pi 3B+).

### Camera

I am using the following type of camera: Network

My camera model is: D-Link 5020L.

### Network Connection

My motionEyeOS unit is connected to the network via: Wifi

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

None

### Log Files

I consider the following log files relevant to this issue:

None. Will provide on request.

==============================================================
Creating stand alone (no network) Pi Cam using either PiZW or Pi3B+.
I create the following files in /boot after imaging SD card with either 20190911 or 20200203:

[ConfigFiles.zip](https://github.com/ccrisan/motioneyeos/files/4173250/ConfigFiles.zip)
dnsmasq.conf, hostapd.conf, os.conf and static_ip.conf
I then boot into the SD card on a PI
This works fine in 0911, but not with 0203. I cannot access the WiFi AP created by 0203 with my phone.
I can access the AP with my phone in 0911.
I can access the web page at 192.168.42.1 running 0911
I can attach a KVM to the Pi and access the command line.
I cannot access via SSH.
I can make the file systems r/w, use the script snippet from /etc/init.d/S60_sshd to generate the keys, run /usr/sbin/sshd manually and start sshd that way, but I've only been able to do it from the KVM.
I found a comment where it's disabled by ThingOS, but not how so it can be reversed? 
Now for the reason WHY (case use):

Need to be able to set date manually for the PI. Need command line access for testing.

or

Under Expert Settings, add option to Date Method, "Direct" and allow input to 'date --set 'YYYY-MM-DD HH-MM-SS' in field below for Date HTTP Host / NTP Server /direct input

Thank you for your consideration.

starbasessd, 2020-02-07:
Sorry, Camera type is Generic USB Camera ( not relevant)

ccrisan, 2020-02-22:
You're right: accessing the board via SSH should be possible in networkless mode.

https://github.com/ccrisan/motioneye/commit/181312e3588a25a132c0ab9dabbe24c674a87fe1 and https://github.com/ccrisan/motioneyeos/commit/0ced6e591d0718b4342c7c28cfcd96685913baba will allow starting of SSHD, FTP and SMB services in networkless mode.

closed 2020-02-22 by ccrisan

starbasessd, 2020-02-22:
> ccrisan/motioneye@181312e
> and 0ced6e5

Awesome, sir, thank you.
If you need tester, let me know.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

ccrisan, 2020-02-22:
@starbasessd I always need testers :) Let's see how the upcoming nightly build behaves.