motioneye-project/motioneyeos #2285: Limit storage per camera

slackwaree, 2020-02-10:
Hello,

So far the only limitation I found in motioneye is to limit the retention time from a camera to minimum 1 day. This might not be good enough. 

Could you please add a storage limit per camera (right now there is none) and when that limit exceeded then it will start deleting the oldest videos.

Right now if the storage gets filled up all what it does is creates 0 byte sized videos.

niko2020, 2020-02-11:
YES! That is a MUST HAVE FEATURE. I was about to post for the same request!
Also, it would be great to have the possibility (checkbox) to automatically overwrite old files if the storage is full (rather than discarding the recording of the new videos).

Mareckil, 2023-11-02:
I am also about this very important feature. And I think not the last one. But seeing that last posts are three years old and nothing happened since then, there's low chance to get it done in some perdictible future. But anyway ;)

bbillp, 2023-11-02:
MotionEye is still working for me on an old Pi-2 and the old 8meg Pi camera capturing up to 2 weeks of data on a 64 gig SD card.  Its sure been a useful App.

Sent from my iPad
>