motioneye-project/motioneyeos #2286: xu4 fan speed

zubzero689, 2020-02-11:
first time using this and i am new to most of what SBC are but i don't like them running hot if i can help it.  thing tend to run better colder so i am wondering if there could be a fan controller for the xu4 because they run hot and the builtin fan curve is for like 60c + i would just like to run that fan 24/7 if i could and all effort to change it with putty/ssh aren't working and i think i understand why but it would be cool if you didn't have to ssh anyway thought it would be a cool feature to have and i didn't see anyone talking about the fan controller and if there way i am sorry

starbasessd, 2020-02-11:
Have you tried:

https://magazine.odroid.com/article/fan-control-tailor-odroid-xu4-perfect-settings/

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

zubzero689, 2020-02-11:
Yes -sh command not found

starbasessd, 2020-02-11:
What command are you giving that your get that result?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

zubzero689, 2020-02-11:
-sh: $: command not found
-sh: sudo: command not found

zubzero689, 2020-02-11:
it works if i take out the sudo 
OMG

starbasessd, 2020-02-11:
Don't use the '$'. It's just a marker. Like '>'
An example command would be:

echo 0 | sudo tee /sys/devices/platform/pwm-fan/hwmon/hwmon0/automatic

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-02-11:
Also, since your are 'admin' you don't need 'sudo'

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

zubzero689, 2020-02-11:
now what is the line to check the temp make sure the fan are working right plz

starbasessd, 2020-02-11:
To set to always on, and full speed:

echo 0 | sudo tee /sys/devices/platform/pwm-fan/hwmon/hwmon0/automatic
echo 255 | sudo tee /sys/devices/platform/pwm-fan/hwmon/hwmon0/pwm1

There are references to scripts and commands, here, but you need to learn a
little programming to get what you want, one of the benefits of SBCs.
https://magazine.odroid.com/article/fan-control-tailor-odroid-xu4-perfect-settings/

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

closed 2020-02-23 by zubzero689