motioneye-project/motioneyeos #2287: camera2 & camera3 fail to record video

kd4kw, 2020-02-12:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

20190911  (0.42)

### Board Model

Raspberry pi zero W
### Camera

RPI remote camera using 5m 

### Network Connection

WiFi and Ethernet

### Peripherals

N/A

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)

I have three pi zero W all running the same version of MotionEyeOs with same model camera.
I am using a RPI 4B as the hub running MotionEye on Raspian current version.
MY question is as I look at the three cameras side by side  on the hub on web site http://x.x.x.x:8765 camera1 will record video and has a little icon in the bottom right corner to take snap shots.
Camera2 and Camera3 do not have this icon and do not record video with motion sensing setup.
All cameras have the exact same setup. 
Is this normal for only camera1 to capture video and not the other two?
Thanks.

starbasessd, 2020-02-12:
Interesting.
My PC based hub has 1 direct attached USB camera and 3 network cameras set
up.
One of my cameras did not have the little blue snapshot icon in the lower
right frame.
I annually checked and compared all and had to turn on 'Enable Manual
Snapshot' under Still Images.
I don't know how it got turned off.

[image: image.png]

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

kd4kw, 2020-02-12:
"Enable Manual Snapshot" is on in all my cameras.

starbasessd, 2020-02-12:
Creating a Pi4 hub...

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-02-12:
Is your PI4 Raspbian Full, Desktop, or Lite?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

kd4kw, 2020-02-12:
Full version 10.

kd4kw, 2020-02-12:
I have no idea what changed or happened but I have the snap photo icon on camera2 now....

starbasessd, 2020-02-12:
Did you lose it on any other?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

kd4kw, 2020-02-12:
OK, I have the icon on all three camera's and can upload to dropbox. It seems if something don't work you turn off that feature and then turn it back on and it might just work then. All of a sudden I have now is the video don't want to play, it is complaining about wrong file type. But that's another issue.

kd4kw, 2020-02-12:
So long and farewell.

closed 2020-02-12 by kd4kw