motioneye-project/motioneyeos #2288: "Show Frame Changes" doesn't work

Treppengelaender, 2020-02-12:
Turning on "Show Frame Changes" in the Motion Detection Section does nothing on motionEyeOS Version 20200203.
There is no red frame around the changed frames. It worked perfectly fine with motionEyeOS  Version 20190911 - but with 20200203 i am using a RTMP stream for the first time.

### motionEyeOS Version

I am running motionEyeOS version: (insert your version here, e.g. 20200203).

### Board Model

I am using the following board/model: (insert your board model here, e.g. Raspberry PI 4).

### Camera

I am using the following type of camera: Network Camera via RTMP

My camera model is: Reolink RLC-420-5MP

### Network Connection

My motionEyeOS unit is connected to the network via: Ethernet

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * none

closed 2020-02-20 by Treppengelaender

TheMrRandomDude, 2020-12-26:
Hello. I also ran into the same issue. How have you managed to solve it?

starbasessd, 2020-12-26:
Which version of motionEyeOS are you using? Which Pi? Try 20190911 (except if you are using a Pi4) or dev20201026.