motioneye-project/motioneyeos #2290: Upgrade fails from 20190119 to 20190911

fredrik-rambris, 2020-02-13:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: 20190119

### Board Model

I am using the following board/model: Raspberry PI 3B+

### Camera

I am using the following type of camera: Network Camera

My camera model is: HikVision

### Network Connection

My motionEyeOS unit is connected to the network via: Ethernet

### Peripherals

I am using the following peripherals that I consider relevant to this issue:
External Hard Disk

### Log Files

I consider the following log files relevant to this issue:

[root@motion .fwupdate]# fwupdate upgrade 20190911
upgrading to 20190911
downloading...
downloaded 20190911
extracting...
not enough space on root partition (230686720 needed, 209715200 available)

 
[root@motion /]# du -sm -h /*
1.7M	/bin
9.0M	/boot
113.2G	/data
0	/dev
3.5M	/etc
112.8G	/home
73.4M	/lib
0	/lib32
43.0K	/libexec
0	/linuxrc
12.0K	/lost+found
1.0K	/media
1.0K	/mnt
2.0K	/opt
0	/proc
1.0K	/root
0	/run
2.4M	/sbin
0	/sys
1.1M	/tmp
90.1M	/usr
3.0K	/var

[root@motion /]# df -h
Filesystem                Size      Used Available Use% Mounted on
/dev/root               193.7M    172.7M      7.0M  96% /
devtmpfs                369.9M         0    369.9M   0% /dev
tmpfs                   374.4M     16.0K    374.4M   0% /dev/shm
tmpfs                   374.4M      1.1M    373.3M   0% /tmp
/dev/mmcblk0p1           29.9M      9.0M     20.9M  30% /boot
/dev/mmcblk0p3            6.2G    432.3M      5.5G   7% /data
/dev/mmcblk0p3            6.2G    432.3M      5.5G   7% /home/ftp/sdcard
/dev/mmcblk0p3            6.2G    432.3M      5.5G   7% /home/ftp/storage
/dev/sda1               916.8G    112.9G    803.8G  12% /data/media/sda1
/dev/sda1               916.8G    112.9G    803.8G  12% /home/ftp/storage/sda1

[root@motion /]# fdisk -l /dev/mmcblk0
Disk /dev/mmcblk0: 7.4 GiB, 7948206080 bytes, 15523840 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0x128c45ba

Device         Boot   Start      End  Sectors  Size Id Type
/dev/mmcblk0p1 *       2048    63487    61440   30M  e W95 FAT16 (LBA)
/dev/mmcblk0p2       204800   614399   409600  200M 83 Linux
/dev/mmcblk0p3      2097152 15523839 13426688  6.4G 83 Linux

starbasessd, 2020-02-13:
"not enough space on root partition (230686720 needed, 209715200
available)"

There have been issues in the past where you cannot upgrade from one
version to the next version, due to partition size requirements or mapping.

You could also try moving videos and pics to your external storage and / or
deleting unneeded ones.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

fredrik-rambris, 2020-02-13:
I have all my media on external drive and not on root partition. I installed 20190119 from image so I have not sized the root partition.

As a side note the error wasn't presented when trying upgrade from GUI. It just completed seemingly as normal and when it was done, nothing had changed.

starbasessd, 2020-02-13:
Checking.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-02-13:
Clean install 8GB SD card
Prior to any attempts at upgrading:
motionEye version 0.39.3
Motion Version 4.1.1
OS Version motionEyeOS 20190119
Run command in SSH connection:   fwupdate upgrade 20190911
[root@meye-d0375a3a ~]# fwupdate upgrade 20190911
upgrading to 20190911
downloading...
downloaded 20190911
extracting...
not enough space on root partition (230686720 needed, 209715200 available)

Update attempt via Web:  Found version 20190911
Install:  Yes
Reports successful update, but shows same versions and didn't reboot
Forced reboot
Post reboot, shows same settings.

At this point, I'd just download the latest version:  20200203, and
re-image. I'd try to do a backup of settings, but there's been some major
settings updates, as well as all the other updates shown between 0119 and
20200203

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-02-13:
Trying something...
Upgrading to 20190427, then 0911...
stage 1 working.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-02-13:
Worked.
Bring system up to 20190427
ssh to pi or use condole
fwupdate upgrade 20190427
allow reboot (automatic)
check Web Interface that it brought it up to motionEyeOS 20190427
ssh to pi or use condole
fwupdate upgrade 20190911
allow reboot (automatic)
check Web Interface that it brought it up to motionEyeOS 20190427

Hope this helps.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

starbasessd, 2020-02-18:
Did upgrading to 20190427 then upgrading to 20190911 work for you?

fredrik-rambris, 2020-02-18:
Just tested it. It worked great.
I noticed that the 20190911 upgrade also output these lines

reallocating root partition...
Partition #2 contains a ext4 signature.

closed 2022-01-19 by starbasessd