motioneye-project/motioneyeos #2292: hi Guys

Rep001, 2020-02-16:
### Preliminary Docs

I confirm that I have read the [CONTRIBUTING](https://github.com/ccrisan/motioneyeos/blob/master/.github/CONTRIBUTING.md) guide before opening this issue.

I confirm that I have read the [FAQ](https://github.com/ccrisan/motioneyeos/wiki/FAQ) before opening this issue.

### motionEyeOS Version

I am running motionEyeOS version: (insert your version here, e.g. 20180314).

### Board Model

I am using the following board/model: (insert your board model here, e.g. Raspberry PI 3B+).

### Camera

I am using the following type of camera: (choose from V4L2, MMAL, Network Camera, Fast Network Camera and Simple MJPEG Camera).

My camera model is: (insert your camera model, e.g. Logitech C290).

### Network Connection

My motionEyeOS unit is connected to the network via: (choose your network connection, e.g. WiFi, Ethernet).

### Peripherals

I am using the following peripherals that I consider relevant to this issue:

 * (e.g. USB WiFi adapter)
 * (e.g. External Hard Disk)

### Log Files

I consider the following log files relevant to this issue:

 * (attach e.g. motioneye.log)
 * (attach e.g. boot.log)

Rep001, 2020-02-16:
hi guys
i am a complete beginner here so please bare with me. I have a Raspi4 and am trying to setup motioneyeos but i cannot get anything to work.
I have formatted the sd card to fat32 and have downloaded the motioneyeos and used etcher to flash to the sd.i have the raspi4 connected to a small screen and all i see is firstly hdmi1 then no signal and the raspi4 appears to just reboot over and over.
any help appreciated...

starbasessd, 2020-02-16:
You didn't answer any of the questions above.
At the minimum, which version of motionEyeOS did you download? You need
https://github.com/ccrisan/motioneyeos/releases/download/20200203/motioneyeos-raspberrypi4-20200203.img.xz
What camera are you going to be using?
How are you going to connect to your network?
If WiFi did you follow the instructions here:
https://github.com/ccrisan/motioneyeos/wiki/Wifi-Preconfiguration

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Rep001, 2020-02-16:
thank you for your reply. i will have a read through before commenting further. cheers

starbasessd, 2020-02-16:
You might also want to check out:
https://groups.google.com/forum/#!forum/motioneye
There are a lot of people willing to help, if you want to give it a try.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

Rep001, 2020-02-16:
very much appreciated thanks. Its a steep learning curve for an oldie but im having fun!
cheers

starbasessd, 2020-02-16:
I'm 60. I do IT Help Desk for a living. Want to learn? I'll help you all
day. Want me to do it for you? Nope.
My setup:
Spectrum ISP to cable modem to Ubiquity EdgeMax Router.
I have several WiFi routers set as Access Points, including DD-WRT Linksys
E1200.
I have 3 D-Link 5020L WiFi/Ethernet cameras and 1 PiZero (running
motionEyeOS) CSI PiCam all the time, and 2 other setups (One
Pi3B+motionEyeOS with attached USB Camera) as test beds.
I have a virtual machine on a Virtualbox Server running Ubuntu 18.10 and
motionEye as my security camera hub. I have a NAS and the VBox Server
acting as storage and backup storage.
I have a Gigabit Ethernet network internally (Fiber and Cat6 copper).
Multiple PCs, ethernet and wireless BlueRay and Tivos, Smart TVs, Game
Consoles, etc, all playing nicely together.
I use my setup to simulate other's setups (and places where I work) to
reproduce issues and find solutions.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?