motioneye-project/motioneyeos #2293: No image on all devices

holzfelix, 2020-02-16:
Hi I have installed the current motioneye os on my raspberrypi 3b+

I have 4 ip cameras. All 4 are fullhd Cameras. All worked fine until I tried to use Motion detection and email notifications ... I activated mail notification on one device. ... now I have this:

[image]

I removed the settings and restarted the rasperry pi ... nothing changed. CPU is at 6%

starbasessd, 2020-02-16:
Did you try deleting and re-set up a single camera to see if that affects
that camera?

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

holzfelix, 2020-02-16:
nothing changed ... still same picture shown ... When I access the cameras webinterface it works properly... only in motionos not ...

starbasessd, 2020-02-16:
Do you have a spare SDCard? put the image on a different card and see if
you can re-build it? If you get the same thing there, try going back to
20190911.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

holzfelix, 2020-02-16:
are there others with this issue on current version?

thats poor. Should I buy a new card? or are there any logs where I can have a look into?

starbasessd, 2020-02-16:
Usually, rebooting a Pi, or rebooting the cameras, or both clear the No
Camera error. There was an issue with Chrome a while back, but it would
show a grey screen for the Camera, not a No Camera.
When you deleted the camera, and re-add it, does it auto-find the type of
camera?
Have you tried connecting to the individual camera stream port (usually
8081-808x) on the Pi? Does that work or same No Camera?
I just tested a 3B+ with 20200203, quick-added 3 IP Network, Simple MJEG,
set motion detect on 1, and Email Notification and worked fine.
Did you change / add admin or user passwords on the Pi?
When I first started testing / supporting, most of the time it was a card
issue, but I have a dozen in the 8GB-128GB range, so it's no big deal to
try a different card for me.
Also, my D-Link cameras are on Jinvoo outlet wifi switches, so easy for me
to restart the cameras.
Are your cameras Ethernet connected, or wifi? How about the Pi?

Please try to answer as many of these questions as you can.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

holzfelix, 2020-02-16:
yes streams are not available as well ... only over their own ones. 

yes autofind typoe works 

All cameras and pi are wired.

I try a new card.

holzfelix, 2020-02-16:
are there cards you can recommend?

starbasessd, 2020-02-16:
I use MicroCenter's store brand mostly. (8GB - 64GB) I do have a couple of
SanDisk Ultra's. If you don't have a network storage device (NAS or SAN)
then I'd recommend something in the 16-32 GB range.  HC or XC Class 10.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

holzfelix, 2020-02-16:
i don't think it is because of the sd map. i had a new one from san disk in there anyway (ran only 2 weeks)

so I deleted the one camera again with motion + email. 

then restarted the raspi. 

Camera 2-4 went then suddenly again there I have been looking forward to .... 

so then I thought okay then I add now the one again. I did it without any problems. But now none of the cameras is working again .... The new camera has got ID 1 again. 

if i delete the camera via the interface, maybe there will be some remains of the camera left somewhere? can i delete the manual delete that it works again? 

would not like to have to do everything again

starbasessd, 2020-02-16:
Can you access the console, either by keyboard and monitor or SSH?
If you can, there is camera files in /data/etc/motion & /data/etc/motioneye
folders.
If you delete all cameras in the web gui, then delete anything that begins
"Camera...conf" in those two folders.
Reboot.
Try again

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

holzfelix, 2020-02-16:
I found a brand new Sandisk 32gb card ... so I reinstalled .. works for the moment

starbasessd, 2020-02-16:
Good deal. Let me know what help you need, if any.

-- 
Thanks

Kevin Shumaker

Personal Tech Support <https://kevinshumaker.wixsite.com/thethirdlevel>

N38° 19' 56.52"
W85° 45' 8.56"

Semper Gumby
“Don't tell people how to do things. Tell them what to do and let them
surprise you with their results.” - G.S. Patton, Gen. USA
Ethics are what we do when no one else is looking.
Quis custodiet ipsos custodes?
“There is no end to the good you can do if you don’t care who
gets the credit.” - C Powell
You know we're sitting on four million pounds of fuel, one nuclear weapon
and a thing that has 270,000 moving parts built by the lowest bidder. Makes
you feel good, doesn't it?

holzfelix, 2020-02-16:
ok I found the issue ... Thats an bug ;) 

so if you make a typo in the network share name ... and then go to save changes ... then the transmission does not work anymore ... not only not the recording 

Thanks for your help

holzfelix, 2020-02-16:
same in 20190911 and in the pre-release 
on safe the test-network share should be automatically triggered